natcl/Artnet #3: Can't manage to receive artnet

reno-, 2015-02-16:
Hi,
I cannot manage to receive artnet with my arduino uno and the neopixel example.

I tried with arduino ip 2.0.0.11 and my laptop sending artnet from 2.0.0.4
I tried the same on submask 10.x.x.x
I cannot see the arduino as a node on the nework, but maybe it's because you support only artnet v1? So I tried to broadcast from ola (as recquired from artnet v1) but it doesn't work better…

so maybe I'm missing something?

thanks for any pointers

natcl, 2015-02-16:
Did you try the more simple examples (artnet receive or artnet receive callback) to see if you are getting any data ?

> I cannot see the arduino as a node on the nework

That's normal, Artpoll reply is not implemented yet.

reno-, 2015-02-16:
thank you @natcl for your quick answer.
Yes, I tried both without results.
with artnet receive example, if I print artnet.read() in the loop, I only have '0'.

I use arduino 1.0.6. maybe I need to compile under arduino 023?

natcl, 2015-02-16:
If it compiles it should be ok, most of the problems I've seen have been on
the client side with a bad configuration.
What did you put as subnet mask and gateway ?
Did you try other ethernet examples to see if you can get the ethernet to
work ?

reno-, 2015-02-16:
ok, i just got it.
OLA must have     
    always_broadcast = true    
and 
    use_limited_broadcast = true    
I only set the first parameter.

Thank you @natcl for your nice tips.

closed 2015-02-16 by reno-