natcl/Artnet #7: Artnet Packets not entirely accurate

roxaskeyheart, 2016-05-14:
((Despite my other issue, I have been doing some tests with some artnet software which seems to work with this library.))

A problem I seem to see with this library is that when ramping a channel to full or zero, the loop reporting the data seems to get cut off pre-maturely sometimes.

Basically a simple example of this is ramping a channel from zero all the way to max (0 -> 255) reasonably quickly, printing out the data every packet received. Basically what happens is you see the data follow the ramp until I hit max ramp but the reported data stops before 255 (Somewhere between 235-253 generally). I think what happens is the reporting stops when I hit max fader/ramp when the library hasn't caught up. Similar when ramping down to 0, the same thing happens in reverse.

It is worth noting that if I ramp slower, the library does indeed reports the full range - there seems to be a certain speed where it becomes a problem, though this is relatively slow still. The main problem is that a lighting console can manipulate values in ramps and other curves at high speeds (let alone manual manipulation on a fader).

Is there a change that can be done to the library to accept listening to data faster OR for a second longer on the top and tail of receiving data?

natcl, 2016-05-14:
What are you using to send the data and what hardware are you using ?

natcl, 2016-05-14:
The Artnet library is as fast as it can be, I have tested this with several thousands of channels.  I suspect your hardware (I assume it's an Arduino Mega from your other issue) is not fast enough.
What ethernet chipset are you using ?

roxaskeyheart, 2016-05-14:
I'll have a look into another ethernet chipset and compare the results. Thanks ^^

natcl, 2016-05-15:
The wiz820io + Teensy 3 combo is known to be quite fast.

closed 2016-11-21 by natcl