natcl/Artnet #9: Wifi / ESP8266

veteze, 2016-06-28:
Hi! I use this library currently to control a dozen universes all connected over ethernet. 2 with arduino controllers and 3 with teensy's.

I've recently gotten some esp8266 wifi modules working. I was wondering if you have any idea if this code would work over wifi? They send data over the serial ports. The particular ones I have receive data on Serial2.

gamecompiler, 2016-09-09:
Yes it works. Got it working now (Needed half hour XD). Will push data "today" (Its kinda late now).

gamecompiler, 2016-09-09:
Oh, rstephan was faster than me:
https://github.com/rstephan/ArtnetWifi

My code is almost the same - there is really no difference. Just changed 1 line in the lib and added WiFi.begin @ the main sketch.

natcl, 2018-02-22:
Now works with ESP8266 and ESP32

closed 2018-02-22 by natcl