natcl/Artnet #10: How to use ArtNet Subnets

Capii1, 2016-10-18:
I'd like to use universes greater than 15 however I can't figure out how to use subnets to assign more universes. Is this possible with this code?

nanu-c, 2016-10-18:
first step, use this sketch: https://github.com/natcl/Artnet/blob/master/examples/ArtnetReceive/ArtnetReceive.ino and look if you can receive packages of an universe higher than 15.

nanu-c, 2016-10-18:
And the best habit is not using universes higher than 15 but change the subnet
http://www.artisticlicence.com/WebSiteMaster/User%20Guides/art-net.pdf page 15

natcl, 2016-10-18:
Yes, it should work with more than 15 universes without any modifications.
Keep in mind this can be a lot of data to process so your micro controller might not be fast enough.

closed 2016-10-18 by natcl