natcl/Artnet #11: 171+ Pixels

Capii1, 2016-10-18:
I've made an LED fixture that works just fine if I only have 170 pixels. The moment I try to run the sketch with the "numberLeds" set at 171 or anything over that, the entire led strand stops working. I'm using the ArtNetNeoPixel sketch to do this. I'm using MadMapper with a single fixture starting on channel 1 of universe 1 with 171+ pixels. I've also tried the LightKey software with no luck going past 170 pixels. 

[image]

natcl, 2016-10-18:
For fun, can you try setting numLeds to a higher value to see if it helps ?
Try 340. I want to see if it's an issue with the maxUniverses calculation

Capii1, 2016-10-18:
I just tried it with 171,200,340,500 and all of them produce the same result.

natcl, 2016-10-18:
I just checked the calculation and it seems ok.  Did you setup MadMapper to send more than 1 universe ?

Capii1, 2016-10-18:
Yes, I've setup it up to send both universes to the same IP address. I'm going to delete the library and re add it to see if that corrects the issue. 

Thanks for the quick response.

natcl, 2016-10-18:
Also on what hardware on you testing this ?
If it doesn't work, please try the following sketch to see what data comes in:
https://github.com/natcl/Artnet/blob/master/examples/ArtnetReceiveCallback/ArtnetReceiveCallback.ino

closed 2016-11-21 by natcl

natcl, 2017-05-16:
> Hello. Sorry but i didn't understand solution to this issue and i've got the same problem.

Can you monitor the data using this and confirm that you receive more than 1 universe ?
http://www.lightjams.com/artnetominator/

lagelat, 2017-05-16:
I'm on OSX... Trying to find an artnet monitor soft.

lagelat, 2017-05-16:
I think that i receive the two universes because all my 280 leds are colored controlled by madmapper but i've got big lags.
Like Capii1, I've setup madmapper to send both universes to the same IP address.

natcl, 2017-05-16:
Can you send me your madmapper project by email ? (without media)

lagelat, 2017-05-16:
This my maddmaper project and a picture of my preferences. It concern the
fixture named LIGNE to universe 6 and 7 (the fixture is a 286 leds  RGB).
The other fixtures (less than 170 leds works perfectly)

This is my sketch with teensy 3.2, WIZ820 and octo as 3v to 5v converter.

/*
This example will receive multiple universes via Artnet and control a strip
of ws2811 leds via
Adafruit's NeoPixel library: https://github.com/adafruit/Adafruit_NeoPixel
This example may be copied under the terms of the MIT license, see the
LICENSE file for details
*/

#include <Artnet.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
#define FASTLED_FORCE_SOFTWARE_SPI
#include "FastLED.h"
//#include <Adafruit_NeoPixel.h>

// Neopixel settings
//const int numLeds = 100; // change for your setup
//const int numberOfChannels = numLeds * 3; // Total number of channels you
want to receive (1 led = 3 channels)
//const byte dataPin = 2;
//Adafruit_NeoPixel leds = Adafruit_NeoPixel(numLeds, dataPin, NEO_GRB +
NEO_KHZ800);

//Fastled settings
#define NUM_LEDS 280
#define DATA_PIN 14
#define CLOCK_PIN 2
const int numberOfChannels = NUM_LEDS * 3; // Total number of channels you
want to receive (1 led = 3 channels)
CRGB leds[NUM_LEDS];

// Artnet settings
Artnet artnet;
const int startUniverse = 6; // 0 pour le n°1

// Check if we got all universes
const int maxUniverses = numberOfChannels / 512 + ((numberOfChannels % 512)
? 1 : 0);
bool universesReceived[maxUniverses];
bool sendFrame = 1;
int previousDataLength = 0;

// Change ip and mac address for your setup
byte ip[] = {192, 168, 0, 118};// 192.168.0.112 pour le n°1
byte mac[] = {0x90, 0xA2, 0xDA, 0x00, 0x44, 0xC7};//0x90, 0xA2, 0xDA, 0x00,
0x44, 0xC1 pour le n°1

void setup()
{

  //Serial.begin(115200);
  artnet.begin(mac, ip);
  FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, BGR, DATA_RATE_MHZ(1)>(leds,
NUM_LEDS);
  //leds.begin();
  initTest();

  // this will be called for each packet received
  artnet.setArtDmxCallback(onDmxFrame);
}

void loop()
{
  // we call the read function inside the loop
  artnet.read();
}

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence,
uint8_t* data)
{
  sendFrame = 1;
  // set brightness of the whole strip
  if (universe == 15)
  {
    //leds.setBrightness(data[0]);
    FastLED.setBrightness(255);
    //leds.show();
    FastLED.show();
  }

  // Store which universe has got in
  if ((universe - startUniverse) < maxUniverses)
    universesReceived[universe - startUniverse] = 1;

  for (int i = 0 ; i < maxUniverses ; i++)
  {
    if (universesReceived[i] == 0)
    {
      //Serial.println("Broke");
      sendFrame = 0;
      break;
    }
  }

  // read universe and put into the right part of the display buffer
  for (int i = 0; i < length / 3; i++)
  {
    int led = i + (universe - startUniverse) * (previousDataLength / 3);
    if (led < NUM_LEDS)
      //leds.setPixelColor(led, data[i * 3], data[i * 3 + 1], data[i * 3 +
2]);Néopixel
      leds[led].setRGB(data[i * 3], data[i * 3 + 1], data[i * 3 + 2]);

  }
  previousDataLength = length;

  if (sendFrame)
  {
    //leds.show();Neopixel
    FastLED.show();
    // Reset universeReceived to 0
    memset(universesReceived, 0, maxUniverses);
  }
}

void initTest()
{
  for (int i = 0 ; i < NUM_LEDS ; i++)
    leds[i].setRGB(127, 0, 0);

  FastLED.show();
  delay(500);
  for (int i = 0 ; i < NUM_LEDS ; i++)
    leds[i].setRGB(0, 127, 0);
  FastLED.show();
  delay(500);
  for (int i = 0 ; i < NUM_LEDS ; i++)
    leds[i].setRGB(0, 0, 127);
  FastLED.show();
  delay(500);
  for (int i = 0 ; i < NUM_LEDS ; i++)
    leds[i].setRGB(0, 0, 0);
  FastLED.show();
}

natcl, 2017-05-16:
Bonjour,
En Artnet les univers commencent à 0 réellement, avez-vous essayé en
mettant startUniverse à 5 ?

Capii1, 2017-05-16:
I'll review my project notes this evening however if I remember correctly my issue was in MadMapper. Can you confirm your artnet settings in madmapper. Are you using unicast, multicast, ect

lagelat, 2017-05-16:
> auth/AbZoR6innCWtmYqLnst-

Non mais j'ai 6 cadres de 170 leds (univers 0 à 5) qui fonctionnent et une
ligne de 286 leds (univers 6 à 7) qui lague.

lagelat, 2017-05-16:
I'm unicast. This is a capture of my MM préférence.

lagelat, 2017-05-16:
The problem concern the univers 6+7, like you i give the same IP adress of
my fixture.

2017-05-16 17:01 GMT+02:00 Cédric Cambon <lagelat@gmail.com>:

> I'm unicast. This is a capture of my MM préférence.
>
>
>> I'll review my project notes this evening however if I remember correctly
>> my issue was in MadMapper. Can you confirm your artnet settings in
>> madmapper. Are you using unicast, multicast, ect
>>
>> <https://github.com/natcl/Artnet/issues/11#issuecomment-301804660>, or mute
>> the thread
>>
>
>
>
> CAMBON Cédric
> 06 61 55 26 78

lagelat, 2017-05-17:
MadMapper fait quelque chose d'étrange quand on fait un fixture de 2
univers, il envoie les valeurs R,V,B  du pixel 170 à leur place (byte n°508
à 510) mais envoie les valeurs du pixel 171 dans l'univers suivant.
Est ce que ta bibliothèque décode ça?
Pour moi il serait logique que le byte 511 et 512 du 1er univers soit le R
et V du pixel 170 et que le B soit le byte 1 de l'univers suivant?

2017-05-16 17:00 GMT+02:00 Cédric Cambon <lagelat@gmail.com>:

> Non mais j'ai 6 cadres de 170 leds (univers 0 à 5) qui fonctionnent et une
> ligne de 286 leds (univers 6 à 7) qui lague.
>
>
>> Bonjour,
>> En Artnet les univers commencent à 0 réellement, avez-vous essayé en
>> mettant startUniverse à 5 ?
>>
>>
>>
>> > This my maddmaper project and a picture of my preferences. It concern
>> the
>> > fixture named LIGNE to universe 6 and 7 (the fixture is a 286 leds RGB).
>> > The other fixtures (less than 170 leds works perfectly)
>> >
>> > This is my sketch with teensy 3.2, WIZ820 and octo as 3v to 5v
>> converter.
>> >
>> > /*
>> > This example will receive multiple universes via Artnet and control a
>> strip
>> > of ws2811 leds via
>> > Adafruit's NeoPixel library: https://github.com/adafruit/Ad
>> afruit_NeoPixel
>> > This example may be copied under the terms of the MIT license, see the
>> > LICENSE file for details
>> > */
>> >
>> > #include <Artnet.h>
>> > #include <Ethernet.h>
>> > #include <EthernetUdp.h>
>> > #include <SPI.h>
>> > #define FASTLED_FORCE_SOFTWARE_SPI
>> > #include "FastLED.h"
>> > //#include <Adafruit_NeoPixel.h>
>> >
>> > // Neopixel settings
>> > //const int numLeds = 100; // change for your setup
>> > //const int numberOfChannels = numLeds * 3; // Total number of channels
>> you
>> > want to receive (1 led = 3 channels)
>> > //const byte dataPin = 2;
>> > //Adafruit_NeoPixel leds = Adafruit_NeoPixel(numLeds, dataPin, NEO_GRB +
>> > NEO_KHZ800);
>> >
>> > //Fastled settings
>> > #define NUM_LEDS 280
>> > #define DATA_PIN 14
>> > #define CLOCK_PIN 2
>> > const int numberOfChannels = NUM_LEDS * 3; // Total number of channels
>> you
>> > want to receive (1 led = 3 channels)
>> > CRGB leds[NUM_LEDS];
>> >
>> >
>> > // Artnet settings
>> > Artnet artnet;
>> > const int startUniverse = 6; // 0 pour le n°1
>> >
>> > // Check if we got all universes
>> > const int maxUniverses = numberOfChannels / 512 + ((numberOfChannels %
>> 512)
>> > ? 1 : 0);
>> > bool universesReceived[maxUniverses];
>> > bool sendFrame = 1;
>> > int previousDataLength = 0;
>> >
>> > // Change ip and mac address for your setup
>> > byte ip[] = {192, 168, 0, 118};// 192.168.0.112 pour le n°1
>> > byte mac[] = {0x90, 0xA2, 0xDA, 0x00, 0x44, 0xC7};//0x90, 0xA2, 0xDA,
>> 0x00,
>> > 0x44, 0xC1 pour le n°1
>> >
>> > void setup()
>> > {
>> >
>> > //Serial.begin(115200);
>> > artnet.begin(mac, ip);
>> > FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, BGR,
>> DATA_RATE_MHZ(1)>(leds,
>> > NUM_LEDS);
>> > //leds.begin();
>> > initTest();
>> >
>> > // this will be called for each packet received
>> > artnet.setArtDmxCallback(onDmxFrame);
>> > }
>> >
>> > void loop()
>> > {
>> > // we call the read function inside the loop
>> > artnet.read();
>> > }
>> >
>> > void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence,
>> > uint8_t* data)
>> > {
>> > sendFrame = 1;
>> > // set brightness of the whole strip
>> > if (universe == 15)
>> > {
>> > //leds.setBrightness(data[0]);
>> > FastLED.setBrightness(255);
>> > //leds.show();
>> > FastLED.show();
>> > }
>> >
>> > // Store which universe has got in
>> > if ((universe - startUniverse) < maxUniverses)
>> > universesReceived[universe - startUniverse] = 1;
>> >
>> > for (int i = 0 ; i < maxUniverses ; i++)
>> > {
>> > if (universesReceived[i] == 0)
>> > {
>> > //Serial.println("Broke");
>> > sendFrame = 0;
>> > break;
>> > }
>> > }
>> >
>> > // read universe and put into the right part of the display buffer
>> > for (int i = 0; i < length / 3; i++)
>> > {
>> > int led = i + (universe - startUniverse) * (previousDataLength / 3);
>> > if (led < NUM_LEDS)
>> > //leds.setPixelColor(led, data[i * 3], data[i * 3 + 1], data[i * 3 +
>> > 2]);Néopixel
>> > leds[led].setRGB(data[i * 3], data[i * 3 + 1], data[i * 3 + 2]);
>> >
>> > }
>> > previousDataLength = length;
>> >
>> > if (sendFrame)
>> > {
>> > //leds.show();Neopixel
>> > FastLED.show();
>> > // Reset universeReceived to 0
>> > memset(universesReceived, 0, maxUniverses);
>> > }
>> > }
>> >
>> > void initTest()
>> > {
>> > for (int i = 0 ; i < NUM_LEDS ; i++)
>> > leds[i].setRGB(127, 0, 0);
>> >
>> > FastLED.show();
>> > delay(500);
>> > for (int i = 0 ; i < NUM_LEDS ; i++)
>> > leds[i].setRGB(0, 127, 0);
>> > FastLED.show();
>> > delay(500);
>> > for (int i = 0 ; i < NUM_LEDS ; i++)
>> > leds[i].setRGB(0, 0, 127);
>> > FastLED.show();
>> > delay(500);
>> > for (int i = 0 ; i < NUM_LEDS ; i++)
>> > leds[i].setRGB(0, 0, 0);
>> > FastLED.show();
>> > }
>> >
>> >
>> > 2017-05-16 14:58 GMT+02:00 Nathanaël Lécaudé <notifications@github.com
>> >:
>> >
>> > > Can you monitor the data using this and confirm that you receive more
>> > than
>> > > 1 universe ?
>> > > http://www.lightjams.com/artnetominator/
>> > >
>> > >
>> > > lecaude.com
>> > > studioimaginaire.com
>> > >
>> :
>> > > >
>> > > > Hello. Sorry but i didn't understand solution to this issue and i've
>> > got
>> > > the same problem.
>> > > >
>> > > >
>> > >
>> or
>> > mute
>> > > the thread
>> AbZoR6innCWtmYqLnst-
>> > c31qybiUXIWSks5r6Z1cgaJpZM4KaPaH>
>> > >
>> >
>> >
>> >
>> > CAMBON Cédric
>> > 06 61 55 26 78
>> >
>> > <https://github.com/natcl/Artnet/issues/11#issuecomment-301801541>, or
>> mute
>> > the thread
>> 6KH1nMV9gk2NaLjyC-vj5GBks5r6bOjgaJpZM4KaPaH>
>>
>> >
>>
>> <https://github.com/natcl/Artnet/issues/11#issuecomment-301802897>, or mute
>> the thread
>>
>
>
>
> CAMBON Cédric
> 06 61 55 26 78

lagelat, 2017-05-17:
Bonjour Nathanaël.
Oublie le mail précédent voici ce dont je viens de m'apercevoir et ce que
je viens de poster sur Madmapper:
By testing i found something.
When i change values of 1st & 2nd Universe at the same time, no lag.
When i change only values of 1st universe, it's lagging.
When i move only values of 2nd universe, it's lagging.

Do MM send the corresponding frame if no values are changed?

I think that the library is testing Universes count and work only mm sends
the two frames of both universes.

Pour moi MM n'envoie que les Univers dont les valeurs changent et ta
bibliothèque demande à avoir tous les univers à chaque trame...
N'est ce pas?

2017-05-17 12:57 GMT+02:00 Cédric Cambon <lagelat@gmail.com>:

> MadMapper fait quelque chose d'étrange quand on fait un fixture de 2
> univers, il envoie les valeurs R,V,B  du pixel 170 à leur place (byte n°508
> à 510) mais envoie les valeurs du pixel 171 dans l'univers suivant.
> Est ce que ta bibliothèque décode ça?
> Pour moi il serait logique que le byte 511 et 512 du 1er univers soit le R
> et V du pixel 170 et que le B soit le byte 1 de l'univers suivant?
>
> 2017-05-16 17:00 GMT+02:00 Cédric Cambon <lagelat@gmail.com>:
>
>> Non mais j'ai 6 cadres de 170 leds (univers 0 à 5) qui fonctionnent et
>> une ligne de 286 leds (univers 6 à 7) qui lague.
>>
>>
>>> Bonjour,
>>> En Artnet les univers commencent à 0 réellement, avez-vous essayé en
>>> mettant startUniverse à 5 ?
>>>
>>>
>>>
>>> > This my maddmaper project and a picture of my preferences. It concern
>>> the
>>> > fixture named LIGNE to universe 6 and 7 (the fixture is a 286 leds
>>> RGB).
>>> > The other fixtures (less than 170 leds works perfectly)
>>> >
>>> > This is my sketch with teensy 3.2, WIZ820 and octo as 3v to 5v
>>> converter.
>>> >
>>> > /*
>>> > This example will receive multiple universes via Artnet and control a
>>> strip
>>> > of ws2811 leds via
>>> > Adafruit's NeoPixel library: https://github.com/adafruit/Ad
>>> afruit_NeoPixel
>>> > This example may be copied under the terms of the MIT license, see the
>>> > LICENSE file for details
>>> > */
>>> >
>>> > #include <Artnet.h>
>>> > #include <Ethernet.h>
>>> > #include <EthernetUdp.h>
>>> > #include <SPI.h>
>>> > #define FASTLED_FORCE_SOFTWARE_SPI
>>> > #include "FastLED.h"
>>> > //#include <Adafruit_NeoPixel.h>
>>> >
>>> > // Neopixel settings
>>> > //const int numLeds = 100; // change for your setup
>>> > //const int numberOfChannels = numLeds * 3; // Total number of
>>> channels you
>>> > want to receive (1 led = 3 channels)
[... 151 lines cut ...]
>>> >
>>> >
>>> > CAMBON Cédric
>>> > 06 61 55 26 78
>>> >
>>> or mute
>>> > the thread
>>> 6KH1nMV9gk2NaLjyC-vj5GBks5r6bOjgaJpZM4KaPaH>
>>>
>>> >
>>>
>>> <https://github.com/natcl/Artnet/issues/11#issuecomment-301802897>, or mute
>>> the thread
>>>
>>
>>
>>
>> CAMBON Cédric
>> 06 61 55 26 78
>>
>
>
>
> CAMBON Cédric
> 06 61 55 26 78

natcl, 2017-05-17:
Oui exact, le sketch s'attend à recevoir tous les univers. Il faudrait faire une version spécifique du sketch pour madmapper, malheureusement je suis sur la route ces jours-ci donc difficile de tester. Tu pourrais toujours enlever le check qui attend de recevoir tous les univers.

lagelat, 2017-05-17:
> 301773838>,

Ah je suis content. Merci. Par contre je sais pas trop comment enlever le
check mais je vais essayer. Si jamais t'as une piste... Si tu n'as pas le
temps de me répondre pas de souci, je sais ce que c'est, je suis souvent en
tournée.
Merci encore pour ton travail.

lagelat, 2017-05-17:
C'est bon j'ai trouvé, je l'utilise dès ce soir. Merci encore.

2017-05-17 16:38 GMT+02:00 Cédric Cambon <lagelat@gmail.com>:

> Ah je suis content. Merci. Par contre je sais pas trop comment enlever le
> check mais je vais essayer. Si jamais t'as une piste... Si tu n'as pas le
> temps de me répondre pas de souci, je sais ce que c'est, je suis souvent en
> tournée.
> Merci encore pour ton travail.
>
>
>> Oui exact, le sketch s'attend à recevoir tous les univers. Il faudrait
>> faire une version spécifique du sketch pour madmapper, malheureusement je
>> suis sur la route ces jours-ci donc difficile de tester. Tu pourrais
>> toujours enlever le check qui attend de recevoir tous les univers.
>>
>>
>> lecaude.com
>> studioimaginaire.com
>>
>> >
>> > Bonjour Nathanaël.
>> > Oublie le mail précédent voici ce dont je viens de m'apercevoir et ce
>> que
>> > je viens de poster sur Madmapper:
>> > By testing i found something.
>> > When i change values of 1st & 2nd Universe at the same time, no lag.
>> > When i change only values of 1st universe, it's lagging.
>> > When i move only values of 2nd universe, it's lagging.
>> >
>> > Do MM send the corresponding frame if no values are changed?
>> >
>> > I think that the library is testing Universes count and work only mm
>> sends
>> > the two frames of both universes.
>> >
>> > Pour moi MM n'envoie que les Univers dont les valeurs changent et ta
>> > bibliothèque demande à avoir tous les univers à chaque trame...
>> > N'est ce pas?
>> >
>> > 2017-05-17 12:57 GMT+02:00 Cédric Cambon <lagelat@gmail.com>:
>> >
>> > > MadMapper fait quelque chose d'étrange quand on fait un fixture de 2
>> > > univers, il envoie les valeurs R,V,B du pixel 170 à leur place (byte
>> n°508
>> > > à 510) mais envoie les valeurs du pixel 171 dans l'univers suivant.
>> > > Est ce que ta bibliothèque décode ça?
>> > > Pour moi il serait logique que le byte 511 et 512 du 1er univers soit
>> le R
>> > > et V du pixel 170 et que le B soit le byte 1 de l'univers suivant?
>> > >
>> > > 2017-05-16 17:00 GMT+02:00 Cédric Cambon <lagelat@gmail.com>:
>> > >
>> > >> Non mais j'ai 6 cadres de 170 leds (univers 0 à 5) qui fonctionnent
>> et
>> > >> une ligne de 286 leds (univers 6 à 7) qui lague.
>> > >>
>> > >> 2017-05-16 16:37 GMT+02:00 Nathanaël Lécaudé <
>> > >>
>> > >>> Bonjour,
>> > >>> En Artnet les univers commencent à 0 réellement, avez-vous essayé en
>> > >>> mettant startUniverse à 5 ?
>> > >>>
>> > >>>
>> > >>>
>> > >>> > This my maddmaper project and a picture of my preferences. It
>> concern
>> > >>> the
>> > >>> > fixture named LIGNE to universe 6 and 7 (the fixture is a 286 leds
>> > >>> RGB).
[... 213 lines cut ...]
>> > >>
>> > >> CAMBON Cédric
>> > >> 06 61 55 26 78
>> > >>
>> > >
>> > >
>> > >
>> > > CAMBON Cédric
>> > > 06 61 55 26 78
>> > >
>> >
>> >
>> >
>> > CAMBON Cédric
>> > 06 61 55 26 78
>> >
>>
>> <https://github.com/natcl/Artnet/issues/11#issuecomment-302102613>, or mute
>> the thread
>>
>
>
>
> CAMBON Cédric
> 06 61 55 26 78

natcl, 2017-05-17:
Super !
Normalement en changeant 
if (sendFrame)
pour if (1) ça devrait aider