natcl/Artnet #13: Checking the Art-Net ID with a length of 9.

rocrail, 2016-11-21:
Checking the Art-Net ID with a length of 9 is not correct because the ID is 7 characters long.
All packets will be dropped because the last 2 bytes of 9 will probably never match.

nanu-c, 2016-11-21:
what do you mean with art-net id?
In the ArtPoll packet definition (http://www.artisticlicence.com/WebSiteMaster/User%20Guides/art-net.pdf) the ID is an

>  Array of 8 characters, the final character is a 
> null termination.
> Value = ‘A’ ‘r’ ‘t’ ‘-‘ ‘N’ ‘e’ ‘t’ 0x00

So I think it's not a bug or something like that

rocrail, 2016-11-21:
the loop is from index zero to kess the 9: 0...8
Index 8 is undefined:
 ‘A’ [0]
 ‘r’ [1]
 ‘t’ [2]
 ‘-‘ [3]
 ‘N’ [4]
 ‘e’ [5]
 ‘t’  [6]
0x00 [7]
 ??? [8]

nanu-c, 2016-11-21:
Then make a pull request please, but for me it was never a problem.

rocrail, 2016-11-21:
Your code:

`      // Check that packetID is "Art-Net" else ignore
      for (byte i = 0 ; i < 9 ; i++)
      {
        if (artnetPacket[i] != ART_NET_ID[i])
          return 0;
      }
`

should be IMHO:

`      // Check that packetID is "Art-Net" else ignore
      for (byte i = 0 ; i < 8 ; i++)
      {
        if (artnetPacket[i] != ART_NET_ID[i])
          return 0;
      }
`

rocrail, 2016-11-21:
Why should I make a pull request? Just change the 9 into a 8...

nanu-c, 2016-11-21:
Actually i am not the owner of this repository so i am not able to change it

rocrail, 2016-11-21:
Attach is a patch.

[fixcheckingid.diff.zip](https://github.com/natcl/Artnet/files/603750/fixcheckingid.diff.zip)

natcl, 2016-11-21:
Thanks.
Fixed by commit 5cc30a4

closed 2016-11-21 by natcl