natcl/Artnet #16: Long Pixel Strands

Capii1, 2017-07-18:
I’ve been using this code with strips between 200-250 pixels flawlessly however I’ve increased the string length to 388 pixels. Once updated to anything over 295 the code no longer works. The initial color flash no longer works and data is no longer sent to the led strip. I tried two different boards on two different strips of 388. I can tell the artnet signal is coming over the Ethernet shield as the indicator lights flash when the data is sent.

Capii1, 2017-07-18:
/*
This example will receive multiple universes via Artnet and control a strip of ws2811 leds via 
Adafruit's NeoPixel library: https://github.com/adafruit/Adafruit_NeoPixel
This example may be copied under the terms of the MIT license, see the LICENSE file for details
*/

#include <Artnet.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
#include <Adafruit_NeoPixel.h>

// Neopixel settings
const int numLeds = 388; // change for your setup
const int numberOfChannels = numLeds * 3; // Total number of channels you want to receive (1 led = 3 channels)
const byte dataPin = 2;
Adafruit_NeoPixel leds = Adafruit_NeoPixel(numLeds, dataPin, NEO_GRB + NEO_KHZ800);

// Artnet settings
Artnet artnet;
const int startUniverse = 1; // CHANGE FOR YOUR SETUP most software this is 1, some software send out artnet first universe as 0.

// Check if we got all universes
const int maxUniverses = numberOfChannels / 512 + ((numberOfChannels % 512) ? 1 : 0);
bool universesReceived[maxUniverses];
bool sendFrame = 1;
int previousDataLength = 0;

// Change ip and mac address for your setup
byte ip[] = {192, 168, 202, 113};
byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x90, 0xEC};

void setup()
{
  Serial.begin(115200);
  artnet.begin(mac, ip);
  leds.begin();
  initTest();

  // this will be called for each packet received
  artnet.setArtDmxCallback(onDmxFrame);
}

void loop()
{
  // we call the read function inside the loop
  artnet.read();
}

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  sendFrame = 1;
  // set brightness of the whole strip 
  if (universe == 15)
  {
    leds.setBrightness(data[0]);
    leds.show();
  }

  // Store which universe has got in
  if ((universe - startUniverse) < maxUniverses)
    universesReceived[universe - startUniverse] = 1;

  for (int i = 0 ; i < maxUniverses ; i++)
  {
    if (universesReceived[i] == 0)
    {
      //Serial.println("Broke");
      sendFrame = 0;
      break;
    }
  }

  // read universe and put into the right part of the display buffer
  for (int i = 0; i < length / 3; i++)
  {
    int led = i + (universe - startUniverse) * (previousDataLength / 3);
    if (led < numLeds)
      leds.setPixelColor(led, data[i * 3], data[i * 3 + 1], data[i * 3 + 2]);
  }
  previousDataLength = length;     
  
  if (sendFrame)
  {
    leds.show();
    // Reset universeReceived to 0
    memset(universesReceived, 0, maxUniverses);
  }
}

void initTest()
{
  for (int i = 0 ; i < numLeds ; i++)
    leds.setPixelColor(i, 127, 0, 0);
  leds.show();
  delay(500);
  for (int i = 0 ; i < numLeds ; i++)
    leds.setPixelColor(i, 0, 127, 0);
  leds.show();
  delay(500);
  for (int i = 0 ; i < numLeds ; i++)
    leds.setPixelColor(i, 0, 0, 127);
  leds.show();
  delay(500);
  for (int i = 0 ; i < numLeds ; i++)
    leds.setPixelColor(i, 0, 0, 0);
  leds.show();
}

Capii1, 2017-07-18:
I've been using MadMapper, xLights, and Jands Vista. I've used these same programs to send to other fixtures with the same number of pixels the only difference is they're being run on the ESP8266 with a different library and those work. That confirms the software is sending the data. I think the issue has something to do with the amount of data the Uno is having to send or process for artnet to cover that many pixels. I've run a color chase sketch from Adafruit and that will cover all the pixels just fine so I'm not sure what is causing the issue.

natcl, 2017-07-18:
Oh this is with a Arduino Uno ?

Capii1, 2017-07-18:
That’s correct. It works just fine up till I set it to receive pixel 296 at that point the whole thing stops working

natcl, 2017-07-18:
The Arduino Uno is quite slow so this sounds like a CPU speed limit. 
I strongly suggest using a faster board like the Teensy 3.2.

Capii1, 2017-07-18:
I have a Mega I will try it on later. the Uno works just fine when spending non artnet data to longer strips of pixels up to 500. Can the teensy connect via Ethernet?

natcl, 2017-07-19:
Yes using the wiz820io adapter. The basic Arduino Ethernet shield is quite slower.

Capii1, 2017-07-19:
So I've tried it with the Mega 2560 and it works just fine. Tested up to 500 pixels. I also tried it with another version of the Uno (DCCduino version) and this version would only work up to 270 pixels. All using the same ethernet shield. My guess is it has to do with the buffer memory.

closed 2017-08-04 by natcl