natcl/Artnet #17: Not able to change network subnet

Olivierdemoustier, 2017-08-04:


natcl, 2017-08-04:
Can you try replacing Artnet.begin in your sketch by this:
```
Ethernet.begin(mac, ip, dns, gateway, subnet); 
```
Let me know if that works...

Olivierdemoustier, 2017-08-04:
Thank you for this quick response, but no Luck,

I tried already by adding this line, but replacing isn’t working either.

This is part of the code.

 

 

byte ip[] = {2, 0, 0, 13};

byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};

byte gateway[] = {2,0,0,254};

byte ddns[] = {2,0,0,250};

byte sub[] = {255,0,0,0};

 

void setup()

{

  Serial.begin(19200);

  Ethernet.begin(mac, ip, ddns, gateway, sub);   // Static

//  artnet.begin(mac, ip);

 

 

Strange is that if I use dns instead of ddns, I get an error….

 

 

Best regards,

     

Olivier Demoustier

+32 475 451 257

 

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

natcl, 2017-08-04:
What error are you getting ? I find it strange your DNS is different than your gateway..

Olivierdemoustier, 2017-08-04:
This is the error,

 

But regarding gateway and dns. It doesn’t matter what value, since I have just 1 computer and 1 arduino in a static environment.

 

I need to be in the 2.xxx.xxx.xxx range, because that’s the official range for ARTNET. And this library only seems to work in the commercial 192.168.XXX.XXX range.

I do think the line artnet.begin(mac, ip);  must be in the setup. Otherwise it willnot start the library?

I was hoping for    artnet .begin(mac, ip, dns, gateway, sub);    would work… :(

 

Best regards,

     

Olivier Demoustier

+32 475 451 257

 

https://www.avast.com/antivirus

natcl, 2017-08-04:
I updated the library so that artnet.begin works with all the parameters. However technically it's exactely the same as calling Ethernet.begin...
I don't see the error in your message.

natcl, 2017-08-04:
You can also try 
```
Ethernet.begin(mac, ip, dns, gateway, sub);
artnet.begin();
```

Olivierdemoustier, 2017-08-04:
I attached a screenshot with the error. But her is what the compiler wrote. 

‘byte dns[]’redeclared as different kind of symbol

 

Thank you so much for your help. :)

Where can I download this new library?

 

Best regards,

     

Olivier Demoustier

+32 475 451 257

 

https://www.avast.com/antivirus

Olivierdemoustier, 2017-08-04:
I tried this, no luck

 

Met vriendelijke groeten,

Best regards,

     

Olivier Demoustier

+32 475 451 257

 

https://www.avast.com/antivirus

Olivierdemoustier, 2017-08-04:
Sorry, tried

Ethernet.begin(mac, ip, dns, gateway, sub);
artnet.begin(mac, ip);

—

 

Met vriendelijke groeten,

Best regards,

     

Olivier Demoustier

+32 475 451 257

 

https://www.avast.com/antivirus

Olivierdemoustier, 2017-08-04:
Thank you so much, this is working

 

byte ip[] = {2, 0, 0, 13};

byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};

byte gateway[] = {2,0,0,254};

byte ddns[] = {2,0,0,254};

byte sub[] = {255,0,0,0};

 

void setup()

{

    Serial.begin(19200);

    Ethernet.begin(mac, ip, ddns, gateway, sub);   // Static

//    artnet.begin(mac, ip);

    artnet.begin();

}

 

Met vriendelijke groeten,

Best regards,

     

Olivier Demoustier

+32 475 451 257

 

https://www.avast.com/antivirus

closed 2017-08-04 by natcl