natcl/Artnet #19: OctoSK6812+Artnet+ 280ledsSK6812 x4: Only first 3 universes ok, next are lagging

lagelat, 2017-08-27:
Hello, everything is in the title.
I tried to change the artnet.h file from #define MAX_BUFFER_ARTNET 530 to #define MAX_BUFFER_ARTNET 1060 restart IDE but nothing better.
Here is a video: [https://youtu.be/trs3NsmK2Xw](url)
Heres is a post on PJRC forum: [https://forum.pjrc.com/threads/46030-OctoSK6812-Artnet-280ledsSK6812-x4-Only-first-3-universes-ok-next-are-lagging](url)
My sketch:`// Receive multiple universes via Artnet and control a strip of ws2811 leds via OctoWS2811
//
// This example may be copied under the terms of the MIT license, see the LICENSE file for details
//  https://github.com/natcl/Artnet
// 
// http://forum.pjrc.com/threads/24688-Artnet-to-OctoWS2811?p=55589&viewfull=1#post55589

/*11  192.168.0.121   0x90, 0xA2, 0xDA, 0x00, 0x44, 0xCB
12  192.168.0.122   0x90, 0xA2, 0xDA, 0x00, 0x44, 0xCC   
*/

#include <Artnet.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
#include <OctoSK6812.h>

// OctoWS2811 settings
const int ledsPerStrip = 280; // change for your setup
const byte numStrips= 4; // change for your setup
DMAMEM int displayMemory[ledsPerStrip*8];
int drawingMemory[ledsPerStrip*8];

OctoSK6812 leds(ledsPerStrip, displayMemory, drawingMemory, SK6812_GRBW);

// Artnet settings
Artnet artnet;
const int startUniverse = 0; // CHANGE FOR YOUR SETUP most software this is 1, some software send out artnet first universe as zero.
const int numberOfChannels = ledsPerStrip * numStrips * 4; // Total number of channels you want to receive (1 led = 4 channels)
byte channelBuffer[numberOfChannels]; // Combined universes into a single array

// Check if we got all universes
const int maxUniverses = numberOfChannels / 512 + ((numberOfChannels % 512) ? 1 : 0);
bool universesReceived[maxUniverses];
bool sendFrame = 1;

// Change ip and mac address for your setup
byte ip[] = {192, 168, 0, 121};
byte mac[] = {0x90, 0xA2, 0xDA, 0x00, 0x44, 0xCB};

void setup()
{
 // Serial.begin(115200);
  artnet.begin(mac, ip);
  leds.begin();
  initTest();

  // this will be called for each packet received
  artnet.setArtDmxCallback(onDmxFrame);
}

void loop()
{
  // we call the read function inside the loop
  artnet.read();
}

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  sendFrame = 1;

  // Store which universe has got in 
  //NOT WORKING WITH MADMAPPER because  MadMapper won't pollute the network with network packets for universes 
  //that have not changed (except one packet each 4 seconds as defined in official ArtNet spec).
  /*if (universe < maxUniverses)
    universesReceived[universe] = 1;

  for (int i = 0 ; i < maxUniverses ; i++)
  {
    if (universesReceived[i] == 0)
    {
      //Serial.println("Broke");
      sendFrame = 0;
      break;
    }
  }
*/
  // read universe and put into the right part of the display buffer
  for (int i = 0 ; i < length ; i++)
  {
    int bufferIndex = i + ((universe - startUniverse) * length);
    if (bufferIndex < numberOfChannels) // to verify
      channelBuffer[bufferIndex] = byte(data[i]);
  }      

  // send to leds
  for (int i = 0; i < ledsPerStrip * numStrips; i++)
  {
    leds.setPixel(i, channelBuffer[(i) * 4], channelBuffer[(i * 4) + 1], channelBuffer[(i * 4) + 2], channelBuffer[(i * 4) + 3]);
  }      
  
  if (sendFrame)
  {
    leds.show();
    // Reset universeReceived to 0
    memset(universesReceived, 0, maxUniverses);
  }
}

void initTest()
{
  for (int i = 0 ; i < ledsPerStrip * numStrips ; i++)
    leds.setPixel(i, 127, 0, 0, 0);
  leds.show();
  delay(500);
  for (int i = 0 ; i < ledsPerStrip * numStrips  ; i++)
    leds.setPixel(i, 0, 127, 0, 0);
  leds.show();
  delay(500);
  for (int i = 0 ; i < ledsPerStrip * numStrips  ; i++)
    leds.setPixel(i, 0, 0, 127,0);
  leds.show();
  delay(500);
  for (int i = 0 ; i < ledsPerStrip * numStrips  ; i++)
    leds.setPixel(i, 0, 0, 0, 127);
  leds.show();
  delay(500);
  for (int i = 0 ; i < ledsPerStrip * numStrips  ; i++)
    leds.setPixel(i, 0, 0, 0, 0);
  leds.show();
}`

natcl, 2017-11-15:
Closing this as was resolved in pjrc thread.
Note: would require implementing artsync

johnmastri, 2018-03-05:
I am having the same issue - I took a look at the pjrc thread and @lagelat ended up using the BlackLED library. Is there no way to leverage your library and have over 3, lag-free universes?

natcl, 2018-03-05:
It's totally possible to have more than 3 lag-free universes, it depends on the software that is used for sending.  What is your exact setup ? That being said supporting Art Sync packets would be nice.  This would be more at the sketch level than the Library level.  I'm a bit busy at the time so would welcome a PR for this.

natcl, 2018-03-06:
I just updated the library with preliminary artsync support.
I have no hardware to test now so it would be great if you could test:
https://github.com/natcl/Artnet/blob/master/examples/ArtnetNeoPixelSync/ArtnetNeoPixelSync.ino
Make sure you read the full thread on pjrc too as you may need to increase buffer sizes in the Artnet library.

natcl, 2018-03-06:
Reopening this issue until I have confimation that artsync works

lukasIO, 2018-05-19:
Hi, 
I had a similiar issue today and started debugging it... I'm sending 10 universes (as 10 udp packets with 530 each), but the universes are received in a rather strange order, I made a new line whenever Universe 0 appears, so the output looks like this

120
120
120
120
120
120
120
120
120
120
120
120
120
120
120
1290
120
1240
1260
1250
1240
1240
1260
1250
1230
12490
1240
1250
1250
1260
1250
1270
1230

to me it seems that this is an issue rather with how Arduino handles the UDP packets... any idea about how to workaround this?

closed 2018-12-04 by natcl