natcl/Artnet #20: Can't get any reply.`

Hantoo, 2017-09-25:
Can't get the arduino mega with ethernet 2 sheild to reply with any serial or get past artnet.begin(mac, ip). If I only include artnet.begin() then it can run past the .begin statement, but does not operate with a mac or ip address. I'm using the standard Ethernet library. Using Ethernet2 library causes errors.

natcl, 2017-09-25:
What kind of errors ?

Hantoo, 2017-09-25:
It doesn't show any errors. It complies and all but it seems that artnet.h file is not happy. I changed part of the Artnet.h to include UIPEthernet files. It then seems to get past the artnet.begin however when I try running the artnetrecieve.ino with the correct IP and Mac, it doesn't return any artnet being broadcasted. I set the arduino to have an IP of 2.0.0.2, and my PC to be 2.0.0.1. And Resolume is broadcasting the artnet data. I've also tried to specify the IP inside resolume, 2.0.0.2.. however it can't seem to find a device on the network with that IP.

Hantoo, 2017-09-26:
Changing artnet.h to include Ethernet2 and EthernetUDP2 fixed issues with ethernet2 module.

closed 2017-09-26 by Hantoo