natcl/Artnet #22: Unicast?

xseignard, 2017-11-14:
Hello!

First of all, thanks for this amazing lib!
I can make it work with madmapper without problem in broadcast mode.

But I tried to configure madmapper in unicast mode without success.

Any idea? Here is my teensy code:
```
#include <Artnet.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
#include <FastLED.h>

// reserved pins by the ethernet and sd adapter
// 4, 8, 9, 10, 11, 12, 13

// number of leds on the strip
const int numLeds = 300;
// 3 dmx channel per led
const int numberOfChannels = numLeds * 3;
// data pin of the strip
const byte dataPin = 2;

// define the array of leds
CRGB leds[numLeds];

// artnet settings
Artnet artnet;
// starting universe
const int startUniverse = 0;

// compute needed universe
const int maxUniverses = numberOfChannels / 512 + ((numberOfChannels % 512) ? 1 : 0);
bool universesReceived[maxUniverses];
bool sendFrame = 1;
int previousDataLength = 0;

// Change ip and mac address for your setup
byte ip[] = {192, 168, 2, 2};
byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};

void setup() {
  initWizAndSD();
  Serial.begin(115200);
  artnet.begin(mac, ip);
  // define the strip here
  FastLED.addLeds<WS2812B, dataPin, RGB>(leds, numLeds);
  initTest();

  // this will be called for each packet received
  artnet.setArtDmxCallback(onDmxFrame);
}

void loop() {
  // we call the read function inside the loop
  artnet.read();
}

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data) {
  Serial.println("FRAME");
  sendFrame = 1;
  // only treat messages related to the universes the node handles
  if ((universe - startUniverse) < maxUniverses) {
    universesReceived[universe - startUniverse] = 1;
  }
  // check if all universes has been received for the current frame
  // if not, do not try to update the pixels (e.g. sendFrame = 0)
  for (int i = 0 ; i < maxUniverses ; i++) {
    if (universesReceived[i] == 0) {
      //Serial.println("Broke");
      sendFrame = 0;
      break;
    }
  }

  // read universe and put into the right part of the display buffer
  for (int i = 0; i < length / 3; i++) {
    int led = i + (universe - startUniverse) * (previousDataLength / 3);
    if (led < numLeds) {
      leds[led] = CRGB(data[i * 3], data[i * 3 + 1], data[i * 3 + 2]);
    }
  }
  previousDataLength = length;

  if (sendFrame) {
    Serial.println("Show");
    FastLED.show();
    // reset universeReceived to 0
    memset(universesReceived, 0, maxUniverses);
  }
}

void initTest() {
  for (int i = 0 ; i < numLeds ; i++)
    leds[i] = CRGB(127, 0, 0);
    FastLED.show();
    delay(500);
    for (int i = 0 ; i < numLeds ; i++) {
      leds[i] = CRGB(0, 127, 0);
    }
    FastLED.show();
    delay(500);
    for (int i = 0 ; i < numLeds ; i++) {
      leds[i] = CRGB(0, 0, 127);
    }
    FastLED.show();
    delay(500);
    for (int i = 0 ; i < numLeds ; i++) {
      leds[i] = CRGB(0, 0, 0);
    }
    FastLED.show();
}

void initWizAndSD() {
  pinMode(9, OUTPUT);
  // begin reset the WIZ820io
  digitalWrite(9, LOW);
  pinMode(10, OUTPUT);
  // de-select WIZ820io
  digitalWrite(10, HIGH);
  pinMode(4, OUTPUT);
  // de-select the SD Card
  digitalWrite(4, HIGH);
  // end reset pulse
  digitalWrite(9, HIGH);
}
```

Regards

natcl, 2017-11-15:
Unicast normally works out of the box, did you try sending with another software ?

xseignard, 2017-11-15:
Hello @natcl, thanks for your answer! I still havent time to check with another software.
I'll try this tomorrow.

Regards

closed 2018-02-23 by natcl