natcl/Artnet #28: Issues with greater than 170 LEDs

JoeSpinogatti, 2018-03-01:
I have a strip of 261 WS2812 LEDs. They are verified to be working correctly with other libraries.

Whenever I try to use the example ARTNETNeopixel.ino, everything works very well with 170 leds. Whenever I update numLeds to something greater than 170, like 171 or 172, LEDs in the strip beyond 170 begin misbehaving. With 171 or 172, the initTest still fires and the strip blinks RGB.

If I update the number to anything greater the 173, the strip does not run the test and does not blink RGB. Any help or suggestions are greatly appreciated.

Thanks!

natcl, 2018-03-01:
What microcontroller are you using ?

JoeSpinogatti, 2018-03-02:
Arduino Uno. 

Chip ATMega328. 2k of SRAM

JoeSpinogatti, 2018-03-02:
Started here: https://mega-byte.nl/controlling-ws2812-led-matrix-ethernet-using-arduino/

natcl, 2018-03-02:
The Arduino Uno is not fast enough to process Artnet packets and drive LEDs at the same time. Do you have access to a Teensy LC or 3.2 ?

JoeSpinogatti, 2018-03-02:
Unfortunately, no. I can only use Arduino Uno. Everything is running at 5v and powered via usb power bricks. 

What I’m confused about it why it breaks precisely at 171+. The article I found talks about driving more than one universe (170 3ch leds) with the code so any idea why it’s breaking?

natcl, 2018-03-02:
Can you try the basic example and see how many universes you get ?

https://github.com/natcl/Artnet/tree/master/examples/ArtnetReceive

nanu-c, 2018-03-02:
There is not enough space for the buffer on the arduino uno. The arduino has 2 KB (ATmega328P) SRAM. I think with 171 the buffer is full. You can use arduino mega (8 KB SRAM) for something like 450 pixels. The teensy has 65536 KB RAM.

annjohnson, 2018-03-06:
Hi

The reason you get errors past 170 pixels is because you have reached the end of your universe. Each pixel is 3 channels, 3×170 is 510... dmx only goes to 512.

Hope this helps!

annjohnson, 2018-03-06:
I'm a bit tired right now, reread the thread and realized you are definitely aware of the info I shared.  Just trying to be helpful! Best of luck.

natcl, 2018-03-06:
Closing this issue as this is not a limitation of the library but the specific hardware used.

closed 2018-03-06 by natcl