natcl/Artnet #29: Compiles but red light on Teensy is on and doesn't show up on network

johnmastri, 2018-03-03:
Any idea what is going on here?

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp: In member function 'uint16_t Artnet::read()':

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:94:40: warning: embedded '\0' in format [-Wformat-contains-nul]

         sprintf((char *)id, "Art-Net\0");

                                        ^

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:94:40: warning: embedded '\0' in format [-Wformat-contains-nul]

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:107:54: warning: embedded '\0' in format [-Wformat-contains-nul]

         sprintf((char *)shortname, "artnet arduino\0");

                                                      ^

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:107:54: warning: embedded '\0' in format [-Wformat-contains-nul]

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:108:64: warning: embedded '\0' in format [-Wformat-contains-nul]

         sprintf((char *)longname, "Art-Net -> Arduino Bridge\0");

                                                                ^

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:108:64: warning: embedded '\0' in format [-Wformat-contains-nul]

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:143:109: warning: embedded '\0' in format [-Wformat-contains-nul]

         sprintf((char *)ArtPollReply.nodereport, "%i DMX output universes active.\0", ArtPollReply.numbports);  

                                                                                                             ^

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:143:109: warning: embedded '\0' in format [-Wformat-contains-nul]

C:\Users\Owner\Documents\Arduino\libraries\Artnet-master\Artnet.cpp:155:1: warning: control reaches end of non-void function [-Wreturn-type]

 }

natcl, 2018-03-03:
Are you able to ping the Teensy ?
What software are you using to send ?

johnmastri, 2018-03-05:
issue has been resolved - not related to the library - thanks!

closed 2018-03-05 by johnmastri