natcl/Artnet #31: SoftwareSerial

pdegior, 2018-03-11:
Is there anyway I can get this working using SoftwareSerial.  I have an LCD that is committed to TX/RX pins. The board I'm using has wifi built in on pins 8 and 9.  This is a Leonardo based board.  I an get software serial working and connected.  I'm just trying to figure out how to get the Artnet Library hooked into it.  Is this something fairly simple to do or should I rather use a Mega with additional serial ports?

natcl, 2018-03-11:
What kind of LCD? How are you sending data to the LCD? Artnet is usually used to control lights and sends streams of channel values so I'm not sure how this translates to an LCD.

pdegior, 2018-03-11:
The LCD/Arduino is from 4DSystems.  It's called the 4Duino.  It's basically a Leonardo with WiFi Chip and the LCD touchscreen is built into it.  We are planning on using the ArtNet signals to control some custom stage equipment.  The touchscreen is being used to control that equipment.  The LCD is interfacing with the Arduino serially.  
Since it occupies pins 0 and 1,  using Serial1 on the Leonardo, they connected the WiFi to pins 8 and 9.  That is why we have to use SoftwareSerial since it is the only way to interface with 2 serial devices.

natcl, 2018-03-11:
Ok, thanks it's clearer now.
The Art-Net library library doesn't need or interfere with the Serial ports.
It uses the standard Ethernet or Wifi stack of your Arduino. You'll have to figure out how to make Wifi work on pins 8 and 9 but this doesn't affect the behaviour of the Artnet library.

pdegior, 2018-03-11:
OK thanks....I'll see if I can figure out how to get them integrated.

closed 2018-12-04 by natcl