natcl/Artnet #32: ESP32sync + SK6812 flickering random colors

Heljick, 2018-04-03:
Hi,
got random colors and flickering with my ESP32 and SK6812 using ArtnetNeoPixelESPSync.
I am sending effects from Touchdesigner and it's working great but I still have random flickering (first led is most of the time green) any idea ?
Power is enough and got GND connected to the ESP board.

Working on wondows 10 and Feather HUZZAH32 by Adafruit, Arduino SDK 1.8.5
Also included those .h

#include <SPI.h>
#include <Artnet.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <Adafruit_NeoPixel.h>

Thanks for your work and for helping !

natcl, 2018-04-03:
Does TouchDesigner support Art-Net 4 / artsync ?

Heljick, 2018-04-03:
yes I am guessing it's Artnet 4 and ArtSync is working aswell

natcl, 2018-04-03:
How many LEDs are you controlling? Does the flicker happen with basic examples from the Neopixel library?

Heljick, 2018-04-03:
Nope, didn't found any Adafruit demo for ESP32 :/ But I can tell it's not a Touchdesigner issue as my effects are working with the good colors. If that can help after initTest() finish I have random LED on instead of all being off. Seen on the web that can be ESP32 related issue which is not working exactly as ESP8622 with leds. Need to foundan other PSU to be sure it's not related but never had issue with thissetup using FadeCandy.

natcl, 2018-04-03:
Is your data line sending to the leds 5V or 3.3V ?

Heljick, 2018-04-03:
The data line is only sending Data, led strip is connected to 5v. I missed a parameter to move from RGB to RGBW, doesn't solve the problem but now after initTest() I always have led 23 on when others are off.

Heljick, 2018-04-03:
Ok so I changed the PSU and redo all the connections, same issue here, seems ESP32 related for me. As with already been test by one of you guys ?

natcl, 2018-04-03:
Yes but the data line is an electrical signal.  These LEDs are designed to run at 5V on the data line and the ESP32 is probably 3.3v which can cause issues.
https://learn.adafruit.com/neopixel-levelshifter/shifting-levels

Heljick, 2018-04-03:
Oh yea didn't seen that coming as I was always working with 5v boards before ! Thanks will have a look and let you know how works ;) thanks

Heljick, 2018-04-06:
So I bought a txb0108 and the issue is still the same :/ I can see that the input data is about 7.5mV and the output data 43mv but the result is the same. LED 23 light up after boot test and flickering during process. Any other idea ?

natcl, 2018-04-06:
Do you have another strip to test?

natcl, 2018-04-06:
> I got some WS2812B, will try it later. Do you know any simple free
> software to test outside touchdesigner ? Can't found good stuff :/

The demos of madmapper and resolume should work

Heljick, 2018-04-06:
Just tried Madmapper demo and got the same glitch. I let know when I will have time to test the WS2812B ;)

Heljick, 2018-04-06:
So with WS2812B and an other power supply I have the same issue, as the first led being green. Colors are ok when static color is asked but glitch appear when colors are moving. Will try some code later to see how it performs. Maybe an issue with the Feather ESP32 but will be wierd :/ As well with WS2812B it's working better without the TXB0108, there is no change on SK6812.

natcl, 2018-04-06:
Would be nice if you could test with only the Neopixel library to see if it's an issue with the network.

Heljick, 2018-04-06:
Nope, it's glitchy as well

Heljick, 2018-04-06:
[image]
https://github.com/adafruit/Adafruit_NeoPixel/issues/139
Seems I am not the only one getting this issue, so I guess It's the board having somthing special which need extra code or modification to run properly.

Heljick, 2018-04-06:
So after some talk with Scott from the other thread this is coming from the wifi connection on the Feather HUZZAH SP32 and works well on ESP8266 only !

closed 2018-12-04 by natcl