natcl/Artnet #34: Any example with MAX/MSP or Touch Designer ?

xseignard, 2018-04-19:
Hello,

I'm looking for examples with MAX/MSP or Touch Designer, sending ArtSync message.

Is there any?

Regards and thanks for your lib!

Heljick, 2018-04-19:
Hey Xavier ! Giveme a couple of minute I will do one for you running on TD ;) I am from Nantes too btw ^^

Heljick, 2018-04-19:
[ArtnetSync_TD.zip](https://github.com/natcl/Artnet/files/1927179/ArtnetSync_TD.zip)
Here it is, inside you can found an example for RGB led and one for RGBW led ;) Enjoy !

xseignard, 2018-04-19:
Hey! That good to see some other "Nantais" here :) Are you still in town? If so we should meet!
Here are the last led stuff I did with my own custom Artnet to led strips controller
https://www.instagram.com/p/Bep1HxNhoiV/

closed 2018-12-04 by natcl