natcl/Artnet #37: Art-Net with single RGB chip

ymatahen, 2018-05-06:
Hi, Im pretty new to this so im not sure how the coding of this should work. I am trying to implement Art-Net onto a PAR style light over wifi with an esp8266. The light is a single RGB setup (no pixels), and i would like the esp to act as the receiver and light the rgb via the pwm pins. Any idea how the sketch might look? 
Thanks.

digitalanimalscollective, 2018-05-07:
Hey!
If I understand you correctly you want to have the data from art net sent out straight to your PWM pins.
That is not too far from the ESP Artnet receive example.
You would need to give an art net address to each of the your PARs, and when looping through the data (either in loop() or inside the callback) Just send the respective data to the PWM pins with digitalWrite or PORT manipulations

I made [a small gist](https://gist.github.com/digitalanimalscollective/411d6a4822fcf11321fc2e9181e4b8e5) from the example I mentioned, hopefully it helps.

verycollective, 2018-05-28:
Fixed the [link](https://gist.github.com/verycollective/411d6a4822fcf11321fc2e9181e4b8e5)
Feel free to get in touch if you need a hand with any specific questions

closed 2018-12-04 by natcl