natcl/Artnet #38: OctoWS2811 with artnet support only 510 leds?

lore463, 2018-08-09:
Hi,

i'm using Artnet example of OctoWS2811.

ledsPerStrip = 300
numStrips = 3

it doesn't work. 
It works fine with less then 510 leds in total (160ledx2strips or 60x8 or 120x4, not 600x1). I'm using teensy 3.5 and i saw that it supports up to 4000 leds.

Is there any problem in the example? maybe in leds.show(); because with 300x3 leds in total the init test go well and if i print the [i] in this code inside the example all goes well:

// read universe and put into the right part of the display buffer
for (int i = 0; i < length / 3; i++)
{
int led = i + (universe - startUniverse) * (previousDataLength / 3);
if (led < numLeds)
leds.setPixel(led, data[i * 3], data[i * 3 + 1], data[i * 3 + 2]);
Serial.print(i);
} 

Data arrive, and all go well until leds.show(); i think...

Any suggest? Thank you

natcl, 2018-08-09:
What software are you using to send ? Are you using the "sync" example ?

lore463, 2018-08-09:
i use max map with imp.artnet externals
i don't know "sync" example, where i can find?

lore463, 2018-08-09:
yes, on max doesn't work sync.

I'm senni universes in serial, but when i use more than 512 leds (e.g. 1536 Byte, 3 universes) the Teensy doesn't work. Or better, i saw in the loop that i write on the first post the data arrive but the leds don't wok. if i use fewer leds all go fine.

starting universe set to 0 either max and teensy, length = 512

natcl, 2018-08-09:
Could you try printing the incoming artnet packets on the teensy side to
see if all is normal?
The Ethernet library on Arduino/Teensy has some buffer settings that
sometimes need to be changed when sending large amounts of data.
For a quick test you could also try replacing line 82 (if (sendFrame)) by
if (1) and see if you get data.

lore463, 2018-08-09:
it works for 5 minutes, later doesn't respond...arg

natcl, 2018-08-09:
That's when you modified the line I told you ? Or are you printing packets coming from Max ?

lore463, 2018-08-10:
yes, but i would use only max msp. Anyway i use only white so the RGB isn't important for me. i'll try the last solution...send to Teensy the same value 0-255 on R-G-B to use it like brightness for white. In this case i'll have 3 universes to run up to 1536 led. i'll change the sketch of OctoWS2811. Or i'll use 2 Teensy.

Anyway there is some problem in OctoWS2811 and imp.dmx externals because is not possible to handle only 3 universes in both.

in Prjc i have 2 posts but there aren't replies!

Very thank you Nat for the support. i'll keep you updated!

Best

closed 2018-12-04 by natcl