natcl/Artnet #39: Is Arduino mini 3.3v @ 8MHz supported?

mrv96, 2018-08-18:
I'm developing a board with Arduino and W5100 for Artnet, can i use 8Mhz Arduino or 8MHz are not enough?

natcl, 2018-08-19:
There's technically no reason it won't work with 8 bit arduinos but I doubt you'll be able to drive alot of channels due to memory limitations...

mrv96, 2018-08-20:
I'm not talking about core bit, but i'm talking about oscillator frequency. Arduino 8 bit usually have 16MHz of clock with 5V as Vcc.

natcl, 2018-08-22:
It still has much less memory than 32 bit boards which is required to buffer larger amounts of artnet universes. As I said it will work in a basic manner but doubt it'll work nicely unless you're ready to optimize lower level libraries...

closed 2018-12-04 by natcl