natcl/Artnet #40: Shouldn't this iterate from 0 to i < 8? Instead of 9?

marcel303, 2018-11-09:
https://github.com/natcl/Artnet/blob/8aa00cc0a82b3c709ccfe6d526d932822e211960/Artnet.cpp#L59

natcl, 2018-11-09:
Normally it should be ok as there is a null character at the end:
#define ART_NET_ID "Art-Net\0"

marcel303, 2018-11-09:
Yes, but iterating from 0 to 9 means 9 bytes are being checked. The ninth byte (at index 8) is the first byte of the opcode. See https://en.wikipedia.org/wiki/Art-Net.

Just starting out with ArtNet and reading up on it, so I may be wrong, but I think the reason it works ok is because the opcode for OpDmx is 0x5000, and the opcode is written low byte first, followed by the high byte. So the 0x00 comes first and then 0x50. It's a zero, but not the zero you think it is. :-) The "Art-Net\0" string has indeed two trailing zeroes so it'll work fine. But should you ever want to handle opcodes with a non-zero low byte it'll break.

natcl, 2018-11-09:
ahh good catch, will fix thanks !

closed 2018-11-09 by natcl

marcel303, 2018-11-09:
Sure, no problem. Thank you for making such an easy to use library available!  I will experiment some more with your code over the weekend.
Mac -> Wifi -> ESP32 -> Dmx serial chip -> Let there be Light. :-)

natcl, 2018-11-09:
No problem ! Let me know how it goes :)

marcel303, 2018-11-12:
Ha @natcl Just writing back to you to let you know it works wonderfully. Works great out of the proverbial box with the ESP32 and the lights I've got! Thanks!

natcl, 2018-11-12:
Excellent ! :)

commit a3186f19 (2018-11-09): Change Art-Net length check fixes #40