natcl/Artnet #41: ArtNet to Neopixel freezing

lyon470, 2018-12-21:
Hello
i have the Artnet Neopixel example running and changed it so that i can set the subnet mask.

Now I have the problem that my output to the neopixels freezes every 4 seconds and nothing changes anymore.

Anybody know what I did wrong?

lyon470, 2018-12-22:
Without the the modification it works correct but i need to configure the subnet

natcl, 2018-12-22:
Is it possible the subnet on the sender is not set correctly?

lyon470, 2018-12-22:
the subnet and the ip on the sender is correct. 
The freezing ist only when the subnet is at 255.0.0.0.

lyon470, 2018-12-22:
Is there another way to change the subnet ?

natcl, 2018-12-22:
Seems like it's part of the Ethernet.begin() constructor:
https://www.arduino.cc/en/Reference/EthernetBegin

lyon470, 2018-12-22:
No i dont't think so because it works correctly when the subnet is at 255.255.255.0.
Only when the subnet is at 255.0.0.0 it is freezing.

Now i figured out that the freezing is every time an Art Pool ist coming in.

natcl, 2018-12-22:
The test sketch you posted above, what is the output in the console? Do you get poll messages and data. What do you mean by freeze ? It stops forever or it's a hiccup in the frame rate?

lyon470, 2018-12-22:
Thanks so much.
Now it's working perfect.

it is possible to change the artnet.begin(mac, ip) to artnet.begin(mac, ip, dns, gateway, subnet);
and how can i change the ArtNode name that is sending in the poll reply

natcl, 2018-12-22:
Yes you can use the Ethernet.begin call as you would in a normal arduino sketch.
You can’t currently change the name via the sketch but you can modify it in the library, here are the lines:
https://github.com/natcl/Artnet/blob/master/Artnet.cpp#L107

lyon470, 2018-12-22:
Thank you for the good support :+1: 

Now everything is working.

closed 2018-12-22 by lyon470