natcl/Artnet #42: Neopixel showing random colors when ArtPoll is receiving

lyon470, 2019-03-05:
Hello,
i have problems with my arduino mega with ethernet shield and resolume sending artnet.
Every time an artpoll is receiving the leds shows shortly an random color.

This is my sketch:

#include <Artnet.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
#include <Adafruit_NeoPixel.h>
#include <EEPROM.h>
#include <Wire.h> 
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27,16,2);
Adafruit_NeoPixel leds = Adafruit_NeoPixel(60, 23, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip = Adafruit_NeoPixel(60, 25, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds2 = Adafruit_NeoPixel(60, 27, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip2 = Adafruit_NeoPixel(60, 31, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds3 = Adafruit_NeoPixel(60, 33, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip3 = Adafruit_NeoPixel(60, 35, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel leds4 = Adafruit_NeoPixel(60, 37, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel strip4 = Adafruit_NeoPixel(60, 39, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel leds5 = Adafruit_NeoPixel(60, 41, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel strip5 = Adafruit_NeoPixel(60, 43, NEO_GRB + NEO_KHZ800);

Artnet artnet;
uint8_t* data;

byte ip[] = {10, 0, 1, 199};
byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};
byte broadcast[] = {10,255,255,255};
byte subnet [] = {255,0,0,0};

void setup() {
  //Serial.begin(9600);
  
  lcd.init();
  lcd.backlight();
  lcd.print("BOOTING....");
  ip[0] = EEPROM.read(1);
  ip[1] = EEPROM.read(2);
  ip[2] = EEPROM.read(3);
  ip[3] = EEPROM.read(4);
  subnet[0] = EEPROM.read(5);
  subnet[1] = EEPROM.read(6);
  subnet[2] = EEPROM.read(7);
  subnet[3] = EEPROM.read(8);
 
  leds.begin();
  strip.begin();
  leds2.begin();
  strip2.begin();
  leds3.begin();
  strip3.begin();
  //leds4.begin();
  //strip4.begin();
  //leds5.begin();
  //strip5.begin();
  initTest();
  artnet.begin(mac,ip);
  Ethernet.setSubnetMask(subnet);
  artnet.setBroadcast(broadcast);
  artnet.setArtDmxCallback(onDmxFrame);
}

void loop() {
data = artnet.getDmxFrame();

for(int i=0;i < 60;i++){
    leds.setPixelColor(i,data[i*3],data[i*3+1],data[i*3+2]);
    strip.setPixelColor(i,data[i*3 + 180],data[i*3+1 + 180],data[i*3+2+ 180]);}

leds.show();
strip.show();
leds2.show();
strip2.show();
leds3.show();
strip3.show();

artnet.read();
}

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence,uint8_t* data, IPAddress remoteIP)
{

}    

void initTest()
{
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 127, 0, 0);
    leds.show();
    strip.setPixelColor(i, 127, 0, 0);
    strip.show();
    leds2.setPixelColor(i, 127, 0, 0);
    leds2.show();
    strip2.setPixelColor(i, 127, 0, 0);
    strip2.show();
    leds3.setPixelColor(i, 127, 0, 0);
    leds3.show();
    strip3.setPixelColor(i, 127, 0, 0);
    strip3.show();}
  delay(500);
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 0, 127, 0);
    leds.show();
    strip.setPixelColor(i, 0, 127, 0);
    strip.show();
    leds2.setPixelColor(i, 0, 127, 0);
    leds2.show();
    strip2.setPixelColor(i, 0, 127, 0);
    strip2.show();
    leds3.setPixelColor(i, 0, 127, 0);
    leds3.show();
    strip3.setPixelColor(i, 0, 127, 0);
    strip3.show();}
  delay(500);
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 0, 0, 127);
    leds.show();
    strip.setPixelColor(i, 0, 0, 127);
    strip.show();
    leds2.setPixelColor(i, 0, 0, 127);
    leds2.show();
    strip2.setPixelColor(i, 0, 0, 127);
    strip2.show();
    leds3.setPixelColor(i, 0, 0, 127);
    leds3.show();
    strip3.setPixelColor(i, 0, 0, 127);
    strip3.show();}
  delay(500);
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 0, 0, 0);
    leds.show();
    strip.setPixelColor(i, 0, 0, 0);
    strip.show();
    leds2.setPixelColor(i, 0, 0, 0);
    leds2.show();
    strip2.setPixelColor(i, 0, 0, 0);
    strip2.show();
    leds3.setPixelColor(i, 0, 0, 0);
    leds3.show();
    strip3.setPixelColor(i, 0, 0, 0);
    strip3.show();}
}

natcl, 2019-03-05:
Does it do the same with an example sketch ?

lyon470, 2019-03-05:
When i put this: 
for(int i=0;i < 60;i++){
leds.setPixelColor(i,data[i3],data[i3+1],data[i3+2]);
strip.setPixelColor(i,data[i3 + 180],data[i3+1 + 180],data[i3+2+ 180]);}
into void onDmxFrame it works correctly.

As soon as i put it into void loop it gets blinking random when an Art Poll is receiving.

lyon470, 2019-03-05:
With the Serial Monitor now i see that the leds is flickering when the OpCode = 8448 and that is the OpPollReply.

natcl, 2019-03-05:
I wonder if your broadcast address is set correctly ?
Are you connected to a router or the mega is connected directly to your computer ?

lyon470, 2019-03-05:
Directly connected to my Computer and the subnet is 255.0.0.0

natcl, 2019-03-05:
You could always try using an older version that doesn't have the ArtPollReplys:
https://github.com/natcl/Artnet/releases
The library was designed for faster microcontrollers (teensy, stm32) so the Mega might not be fast enough/have enough memory.

natcl, 2019-03-05:
Also why are you mixing loop the onDmxFrame frame ?
If you are using the callback loop should only have
 `artnet.read();`

lyon470, 2019-03-05:
I don't understand why the speed are important by this error.
Always when an ArtPoll Reply is receiving from the arduino, the data if from onDmxFrame or from artnet.getDmxFrame is random.

How got to the broadcast adress when:
Arduino: 10.0.1.99 255.0.0.0
PC: 10.0.1.2 255.0.0.0

natcl, 2019-03-05:
Can you try version 1 and see it it does the same ?

natcl, 2019-03-05:
Actually the code you posted is wrong.
if you use the loop you need to check if the packet your received is ART_DMX which you're not doing.
Here's a corrected version:
```
#include <Artnet.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
#include <Adafruit_NeoPixel.h>
#include <EEPROM.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16, 2);
Adafruit_NeoPixel leds = Adafruit_NeoPixel(60, 23, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip = Adafruit_NeoPixel(60, 25, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds2 = Adafruit_NeoPixel(60, 27, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip2 = Adafruit_NeoPixel(60, 31, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds3 = Adafruit_NeoPixel(60, 33, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip3 = Adafruit_NeoPixel(60, 35, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel leds4 = Adafruit_NeoPixel(60, 37, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel strip4 = Adafruit_NeoPixel(60, 39, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel leds5 = Adafruit_NeoPixel(60, 41, NEO_GRB + NEO_KHZ800);
//Adafruit_NeoPixel strip5 = Adafruit_NeoPixel(60, 43, NEO_GRB + NEO_KHZ800);

Artnet artnet;
uint8_t *data;

byte ip[] = {10, 0, 1, 199};
byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};
byte broadcast[] = {10, 255, 255, 255};
byte subnet[] = {255, 0, 0, 0};

void setup()
{
  //Serial.begin(9600);

  lcd.init();
  lcd.backlight();
  lcd.print("BOOTING....");
  ip[0] = EEPROM.read(1);
  ip[1] = EEPROM.read(2);
  ip[2] = EEPROM.read(3);
  ip[3] = EEPROM.read(4);
  subnet[0] = EEPROM.read(5);
  subnet[1] = EEPROM.read(6);
  subnet[2] = EEPROM.read(7);
  subnet[3] = EEPROM.read(8);

  leds.begin();
  strip.begin();
  leds2.begin();
  strip2.begin();
  leds3.begin();
  strip3.begin();
  //leds4.begin();
  //strip4.begin();
  //leds5.begin();
  //strip5.begin();
  initTest();
  artnet.begin(mac, ip);
  Ethernet.setSubnetMask(subnet);
  artnet.setBroadcast(broadcast);
}

void loop()
{
  uint16_t r = artnet.read();
  if (r == ART_DMX)
  {
    data = artnet.getDmxFrame();
    for (int i = 0; i < 60; i++)
    {
      leds.setPixelColor(i, data[i3], data[i3 + 1], data[i3 + 2]);
      strip.setPixelColor(i, data[i3 + 180], data[i3 + 1 + 180], data[i3 + 2 + 180]);
    }

    leds.show();
    strip.show();
    leds2.show();
    strip2.show();
    leds3.show();
    strip3.show();
  }
}

void initTest()
{
  for (int i = 0; i < 170; i++)
  {
    leds.setPixelColor(i, 127, 0, 0);
    leds.show();
    strip.setPixelColor(i, 127, 0, 0);
    strip.show();
    leds2.setPixelColor(i, 127, 0, 0);
    leds2.show();
    strip2.setPixelColor(i, 127, 0, 0);
    strip2.show();
    leds3.setPixelColor(i, 127, 0, 0);
    leds3.show();
    strip3.setPixelColor(i, 127, 0, 0);
    strip3.show();
  }
  delay(500);
  for (int i = 0; i < 170; i++)
  {
    leds.setPixelColor(i, 0, 127, 0);
    leds.show();
    strip.setPixelColor(i, 0, 127, 0);
    strip.show();
    leds2.setPixelColor(i, 0, 127, 0);
    leds2.show();
    strip2.setPixelColor(i, 0, 127, 0);
    strip2.show();
    leds3.setPixelColor(i, 0, 127, 0);
    leds3.show();
    strip3.setPixelColor(i, 0, 127, 0);
    strip3.show();
  }
  delay(500);
  for (int i = 0; i < 170; i++)
  {
    leds.setPixelColor(i, 0, 0, 127);
    leds.show();
    strip.setPixelColor(i, 0, 0, 127);
    strip.show();
    leds2.setPixelColor(i, 0, 0, 127);
    leds2.show();
    strip2.setPixelColor(i, 0, 0, 127);
    strip2.show();
    leds3.setPixelColor(i, 0, 0, 127);
    leds3.show();
    strip3.setPixelColor(i, 0, 0, 127);
    strip3.show();
  }
  delay(500);
  for (int i = 0; i < 170; i++)
  {
    leds.setPixelColor(i, 0, 0, 0);
    leds.show();
    strip.setPixelColor(i, 0, 0, 0);
    strip.show();
    leds2.setPixelColor(i, 0, 0, 0);
    leds2.show();
    strip2.setPixelColor(i, 0, 0, 0);
    strip2.show();
    leds3.setPixelColor(i, 0, 0, 0);
    leds3.show();
    strip3.setPixelColor(i, 0, 0, 0);
    strip3.show();
  }
}
```

dkaulukukui, 2019-03-11:
I’ve had issues with neo pixels acting up if you try to update them too fast. I found anything >50hz causes it to happen. Maybe try add a small delay to your loop. 20ms or so.

natcl, 2019-03-11:
It's not a delay problem in that case it's really that the art_poll packets weren't filtered so were appearing on the strip. @lyon470 did my updated code fix your issue ?

lyon470, 2019-03-11:
Yes the updated code fixed my issue but in can only receive 3 universes.
Do i have an issue in my sketch or are the library only for 3 universes or is the arduino mega to slow for 4 universes ?

natcl, 2019-03-11:
The library can handle more than 3 universes.  The mega might indeed be too slow or lack memory for more than 3 universes, I only tested it on Teensy and ESP32 which has a much faster CPU and more memory.

natcl, 2019-03-11:
I've tested 8 on my side but I think it can handle more.

lyon470, 2019-03-13:
Can i use my Arduino ethernet shield with an teensy 3.2 ?

lyon470, 2019-03-14:
Have anybody an idea ? 
i don't become connection between my ardunio ethernet shield 2 and my teensy 3.2 ?

natcl, 2019-03-14:
It should work providing you find the correct pinouts

lyon470, 2019-03-14:
After the whole day to find out the correct pinout i have the solution and it works now.
I have upload the same sketch (i used with arduino mega before) into the Teensy but i don't get ArtPoll Replys. This means Resolume don't show me the Teensy as "ArtNet Node".
The settings from network are the same.

closed 2019-03-30 by lyon470