natcl/Artnet #44: How many Universe ?

lyon470, 2019-04-07:
Hello,
I have an Teensy 3.2 and 3.6 how many leds/universe can i drive per Teensy ?

natcl, 2019-04-07:
I sadly don't have an official number as I haven't been able to test the upper limits. It should normally be able to drive a decent amount (4-5 minimum I would think) there's some tweaks that can be done to the Ethernet library to improve performance too. If you search for Artnet on the pjrc forums you'll find some posts about this.

lyon470, 2019-04-07:
I have the teensy 3.2 and 5 universe Input but the last universe is lacking.
Have i an issue in my code or is this the limit ?

natcl, 2019-04-07:
I'm pretty sure some people were able to drive more that 5 universes, try searching the PJRC forums for more info. It depends on a lot of things, ethernet library tweaking, type of LEDs you are driving...
At some point I remember lowering throughput of the ethernet connection on the computer to 10mbps speed up things.

lyon470, 2019-04-07:
I don't find anything in the forum.
Do you have an link ?

natcl, 2019-04-07:
Not a specific one but this search should help:
https://forum.pjrc.com/search.php?searchid=4622601

natcl, 2019-04-07:
And this thread has more specific info:
https://forum.pjrc.com/threads/28975-wiz820io-max-speed?highlight=Artnet+universes

lyon470, 2019-04-07:
i have the Ethernet Shield with the W5500 chip.
Is this a Problem ?

natcl, 2019-04-07:
I'm not sure, I only test with the wiz820io (
https://www.wiznet.io/product-item/wiz820io/) on my side...

lyon470, 2019-04-07:
I want to use the Ethernet2 lib but i get an error.
what do i have to Change ?

natcl, 2019-04-07:
I'm not familiar with the Ethernet2 lib, sorry...

lyon470, 2019-04-07:
Ethernet2 lib is specifically for the W5500 chip

natcl, 2019-04-07:
According to this thread, the standard Ethernet library of the Teensy
supports the W5500 chip:
https://forum.pjrc.com/threads/28642-Wiznet-W5500-support

lyon470, 2019-04-07:
Yes but why i got only 4 universe incoming?
I see in the Serial monitor that universe 1-4 are perfect but i don't get more than that

natcl, 2019-04-07:
It could be for various reasons, the fact the wiz5500 is supported by
Teensy's Ethernet library doens't mean it's optimized, you should also
check the different optimization tricks I posted in the threads above.
Posting your code would help too.

lyon470, 2019-04-08:
Yeah many thanks i tested it now with 13 universes without lacking.

But now i have another Problem/issue.
Resolume shows me all artnet nodes in my Network, also the Arduino.
As soon as i tell resolume to send unicast the Arduino isn't anymore in the artnet node list.
When i now change back to broadcast ,the Arduino receives the artnet but isn't in the artnet node list from resolume.

I use the one socket lib from the pjrc Forum.
Hope you can help me.

natcl, 2019-04-09:
Not sure what you mean by socket lib ?

lyon470, 2019-04-13:
After a few long days it now work.
I tested 13 Universes but it can be more.

But now i have another problem.
I have two arduino teensy in my network and send artnet over resolume.
In broadcast mode from resolume it work very good but when i change the artnet to unicast the led stripes breaks every 2 seconds for about 1 second.
This is only in unicast and nothing other is changed.
When i now turn off one teensy it work also in unicast mode but not with two teensys.

Why ??
Have anybody an idea ?

natcl, 2019-04-13:
You'll have to post your code and detailed network setup.

lyon470, 2019-04-13:
Resolume(PC) - 10.0.1.7  255.0.0.0
Teensy 1 - 10.0.1.99  255.0.0.0
Teensy 2 - 10.0.1.100  255.0.0.0

Code:
#include <Artnet.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>
#include <Adafruit_NeoPixel.h>
#include <Wire.h> 
#include <EEPROM.h>

Adafruit_NeoPixel leds = Adafruit_NeoPixel(170, 2, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip = Adafruit_NeoPixel(170, 3, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds2 = Adafruit_NeoPixel(170, 4, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip2 = Adafruit_NeoPixel(170, 5, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds3 = Adafruit_NeoPixel(170, 6, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip3 = Adafruit_NeoPixel(170, 7, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds4 = Adafruit_NeoPixel(170, 8, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip4 = Adafruit_NeoPixel(170, 9, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel leds5 = Adafruit_NeoPixel(170, 14, NEO_GRB + NEO_KHZ800);

Artnet artnet;
uint8_t *data;

byte ip[] = {10, 0, 1, 100};
byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};
byte broadcast[] = {10,255,255,255};
byte subnet [] = {255,0,0,0};
byte dns [] = {};
byte gateway [] = {};

int first_universe = 10;

void setup() {
  Serial.begin(9600);

  leds.begin();
  strip.begin();
  leds2.begin();
  strip2.begin();
  leds3.begin();
  strip3.begin();
  leds4.begin();
  strip4.begin();
  leds5.begin();
  initTest();
  Ethernet.begin(mac, ip, dns, gateway, subnet);
  artnet.begin();
  
  artnet.setBroadcast(broadcast);

}

void loop() { 
  
uint16_t r = artnet.read();
uint16_t uni = artnet.getUniverse();
  
  if (r == ART_DMX && uni == first_universe){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    leds.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    leds.show();}

 if (r == ART_DMX && uni == first_universe + 1){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    strip.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    strip.show();}

  if (r == ART_DMX && uni == first_universe + 2){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    leds2.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    leds2.show();}

  if (r == ART_DMX && uni == first_universe + 3){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    strip2.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    strip2.show();}

 if (r == ART_DMX && uni == first_universe + 4){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    leds3.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    leds3.show();}

 if (r == ART_DMX && uni == first_universe + 5){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    strip3.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    strip3.show();}

 if (r == ART_DMX && uni == first_universe + 6){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    leds4.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    leds4.show();}

if (r == ART_DMX && uni == first_universe + 7){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    strip4.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    strip4.show();}

 if (r == ART_DMX && uni == first_universe + 8){
    data = artnet.getDmxFrame();
    for (int i = 0; i < 170; i++){
    leds5.setPixelColor(i, data[i*3], data[i*3 + 1], data[i*3 + 2]);}
    leds5.show();}

}

void initTest()
{
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 127, 0, 0);
    leds.show();
    strip.setPixelColor(i, 127, 0, 0);
    strip.show();
    leds2.setPixelColor(i, 127, 0, 0);
    leds2.show();
    strip2.setPixelColor(i, 127, 0, 0);
    strip2.show();
    leds3.setPixelColor(i, 127, 0, 0);
    leds3.show();
    strip3.setPixelColor(i, 127, 0, 0);
    strip3.show();
    leds4.setPixelColor(i, 127, 0, 0);
    leds4.show();
    strip4.setPixelColor(i, 127, 0, 0);
    strip4.show();
    leds5.setPixelColor(i, 127, 0, 0);
    leds5.show();}
  delay(300);
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 0, 127, 0);
    leds.show();
    strip.setPixelColor(i, 0, 127, 0);
    strip.show();
    leds2.setPixelColor(i, 0, 127, 0);
    leds2.show();
    strip2.setPixelColor(i, 0, 127, 0);
    strip2.show();
    leds3.setPixelColor(i, 0, 127, 0);
    leds3.show();
    strip3.setPixelColor(i, 0, 127, 0);
    strip3.show();
    leds4.setPixelColor(i, 0, 127, 0);
    leds4.show();
    strip4.setPixelColor(i, 0, 127, 0);
    strip4.show();
    leds5.setPixelColor(i, 0, 127, 0);
    leds5.show();}
  delay(300);
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 0, 0, 127);
    leds.show();
    strip.setPixelColor(i, 0, 0, 127);
    strip.show();
    leds2.setPixelColor(i, 0, 0, 127);
    leds2.show();
    strip2.setPixelColor(i, 0, 0, 127);
    strip2.show();
    leds3.setPixelColor(i, 0, 0, 127);
    leds3.show();
    strip3.setPixelColor(i, 0, 0, 127);
    strip3.show();
    leds4.setPixelColor(i, 0, 0, 127);
    leds4.show();
    strip4.setPixelColor(i, 0, 0, 127);
    strip4.show();
    leds5.setPixelColor(i, 0, 0, 127);
    leds5.show();}
  delay(300);
  for (int i = 0 ; i < 170 ; i++){
    leds.setPixelColor(i, 0, 0, 0);
    leds.show();
    strip.setPixelColor(i, 0, 0, 0);
    strip.show();
    leds2.setPixelColor(i, 0, 0, 0);
    leds2.show();
    strip2.setPixelColor(i, 0, 0, 0);
    strip2.show();
    leds3.setPixelColor(i, 0, 0, 0);
    leds3.show();
    strip3.setPixelColor(i, 0, 0, 0);
    strip3.show();
    leds4.setPixelColor(i, 0, 0, 0);
    leds4.show();
    strip4.setPixelColor(i, 0, 0, 0);
    strip4.show();
    leds5.setPixelColor(i, 0, 0, 0);
    leds5.show();}
}

lyon470, 2019-04-17:
I have now changed to the OctoWS2811 library and this worked perfect but suddeny it started to lack and now no matter what i do it doesn't stop to lack.
I use the demo Sketch artnet_octows2811 and only changed ip and the leds.

natcl, 2019-04-17:
But it worked fine at some point ?
Anything in your network sending alot of data?

MathieuMH, 2020-01-14:
@lyon470 have you tried the fasted library? It looks very promising since it is more efficient per pixel. This should allow more data to be processed.

xseignard, 2020-04-06:
Hello,
Teensy 3.2 + octows2811 lib is the best match for teensy based artnet pixel controller.
Few things to do/respect for improved perfs:
- use w5500!!
- use UNICAST!!
- reduce the `MAX_SOCK_NUM` to 1, e.g. replace the following higlighted code: https://github.com/PaulStoffregen/Ethernet/blob/master/src/Ethernet.h#L36-L40
by:
```
#define MAX_SOCK_NUM 1
```
- enable `ETHERNET_LARGE_BUFFERS `, e.g. uncomment the following line: https://github.com/PaulStoffregen/Ethernet/blob/master/src/Ethernet.h#L48
- increase SPI clock speed, e.g. comment the following line (https://github.com/PaulStoffregen/Ethernet/blob/master/src/utility/w5100.h#L21) and uncomment this one (https://github.com/PaulStoffregen/Ethernet/blob/master/src/utility/w5100.h#L28)

Now you have increased the perfs of the `Ethernet` lib. After that there's few more tricks on the `octows2811` lib you can do if you don't need a drawing buffer (it should be the case if you exclusively controls the leds from resolume)

```
OctoWS2811 leds(MAX_NUM_LED_PER_OUTPUT, displayMemory, NULL, ledConfig);
```

Like this you'd be able to control 8 outs @ 44fps on teensy 3.2

manuelgonzalvez, 2020-05-13:
Hi @xseignard , thanks for your tips. do you have any working example for this?.

 When you say 
> Like this you'd be able to control 8 outs @ 44fps on teensy 3.2

8 universes?  i need about 13-16. is this possible? 

Thanks in advance.

McAleerElk, 2022-10-17:
> 

Did you ever get a reply m8? Or did you test out how many universes you could drive per output?

McAleerElk, 2022-10-17:
> Hi @xseignard , thanks for your tips. do you have any working example for this?.
> 
> When you say
> 
> > Like this you'd be able to control 8 outs @ 44fps on teensy 3.2
> 
> 8 universes? i need about 13-16. is this possible?
> 
> Thanks in advance.

Did you ever get a reply m8? Or did you test out how many universes you could drive per output?

McAleerElk, 2022-10-17:
> Yeah many thanks i tested it now with 13 universes without lacking.
>
> But now i have another Problem/issue. Resolume shows me all artnet nodes in my Network, also the Arduino. As soon as i tell [...]

Is this from one single output?

As I'm aware, the LED itself becomes a bottleneck, as it can only parse so much data, if you want to uphold a given FPS (30Hz or any other number). It receives PWM, bites off the first 3 or 4 bytes (RGB or RGBW), and outputs the remaining part of the signal. But - if you try to feed it 8 universes of PWM signal, it cannot parse it fast enough, and this delay with stack for each LED - adding latency/lagg or similar.

Does anyone experience this problem? Is it not a problem in practice? 

8 universes is 27 meters of LED if RGB-pixels are used... That's quite a lot. And you can have multiple outputs simultaneously?

joex92, 2024-02-13:
Hi,
I got a Teensy4.1 with the ethernet kit...
tried the receiver example and I'm getting up to 4 universes and a half (i get data up to 2304 channels)
how do I get more universes?

natcl, 2024-02-13:
What example are you using ?

joex92, 2024-02-13:
@natcl  this one: https://github.com/natcl/Artnet/blob/master/examples/Basic/ArtnetReceive/ArtnetReceive.ino
I made some minor changes:
```C
/*
This is a basic example that will print out the header and the content of an ArtDmx packet.
This example uses the read() function and the different getter functions to read the data.
This example may be copied under the terms of the MIT license, see the LICENSE file for details
*/

#include <Artnet.h>
#include <NativeEthernet.h>
#include <NativeEthernetUdp.h>
#include <SPI.h>

Artnet artnet;

// Change ip and mac address for your setup
byte ip[] = {2, 0, 0, 2};
byte broadcast[] = {2, 255, 255, 255};
byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};

void setup()
{
  Serial.begin(115200);
  Serial.println("Starting...");
  artnet.begin(mac, ip);
  artnet.setBroadcast(broadcast);
}

void loop()
{
  uint16_t r = artnet.read();
  if(r == ART_POLL)
  {
    Serial.println("POLL");
  }
  if (r == ART_DMX)
  {
    // print out our data
    Serial.print("universe number = ");
    Serial.print(artnet.getUniverse());
    Serial.print("\tdata length = ");
    Serial.print(artnet.getLength());
    Serial.print("\tsequence n0. = ");
    Serial.println(artnet.getSequence());
    Serial.print("DMX data: ");
    for (int i = 0 ; i < artnet.getLength() ; i++)
    {
      const int ch = i + artnet.getUniverse() * 512;
      if ( ch < 10 )
        Serial.print(" ");
      if ( ch < 100 )
        Serial.print(" ");
      if ( ch < 1000 )
        Serial.print(" ");
      Serial.print(ch);
      Serial.print(":");
      const int v = artnet.getDmxFrame()[i];
      if ( v < 10 )
        Serial.print(" ");
      if ( v < 100 )
        Serial.print(" ");
      Serial.print(v);
      Serial.print("  ");
    }
    Serial.println();
    Serial.println();
  }
}
```

I had to change `Ethernet.h` and `EthernetUDP.h` to `NativeEthernet.h` and `NativeEthernetUdp.h` for this code to work...
I'm just printing the data,

joex92, 2024-02-13:
I added some spaces depending on the values for better reading... 
thing is that when I try sending more data, like a fixture of more than 2304 channels, data is lost...
for example, I set a LRGB fixture of 128px12p, and only getting the top 128x4 but occasionally getting the 11th row or the 4th row is incomplete... it's weird...

natcl, 2024-02-13:
There's nothing in the library preventing getting more universes, normally if the board is receiving it it should work unless there is a network configuration issue...

joex92, 2024-02-13:
Then why can't the teensy get all the data from a big fixture?

joex92, 2024-02-13:
is there a hardware limitation on the Ethernet module of the Teensy?

joex92, 2024-02-14:
that's another weird thing... when doing the testing, I'm only sending 1 frame at a time... like I just send 1 frame and it still not everything gets there...
another thing I was thinking is that would it be a bandwidth  issue? like it's too fast? how can I decrease the bandwidth?
would using an old 100Mb switch in between work?

natcl, 2024-02-14:
I know that with the older Teensy 3.2 forcing the Ethernet speed to 10mbps in the OS would help. It might also be a buffer size issue. You'll need to search on the pjrc forums as this is more an Ethernet issue than an Artnet issue, if you go back on this thread there are a couple of links to the pjrc forum with Ethernet optimiza tips.

joex92, 2024-02-14:
> Hello, Teensy 3.2 + octows2811 lib is the best match for teensy based artnet pixel controller (thanks to the DMA buffers, not sure fastled would bring any improvements in a case of artnet only controller).
> 
> Few things to do/respect for improved perfs:
> 
> * use w5500!! because of its larger i/o buffers
> * use UNICAST!!
> * reduce the `MAX_SOCK_NUM` to 1, e.g. replace the following highlighted code: https://github.com/PaulStoffregen/Ethernet/blob/master/src/Ethernet.h#L36-L40
>   by:
> 
> ```
> #define MAX_SOCK_NUM 1
> ```
> 
> * enable `ETHERNET_LARGE_BUFFERS `, e.g. uncomment the following line: https://github.com/PaulStoffregen/Ethernet/blob/master/src/Ethernet.h#L48
> * increase SPI clock speed, e.g. comment the following line (https://github.com/PaulStoffregen/Ethernet/blob/master/src/utility/w5100.h#L21) and uncomment this one (https://github.com/PaulStoffregen/Ethernet/blob/master/src/utility/w5100.h#L28)
> 
> Now you have increased the perfs of the `Ethernet` lib. After that there's few more tricks on the `octows2811` lib you can do if you don't need a drawing buffer (it should be the case if you exclusively controls the leds from resolume)
> 
> ```
> OctoWS2811 leds(MAX_NUM_LED_PER_OUTPUT, displayMemory, NULL, ledConfig);
> ```
> 
> Like this you'd be able to control 8 outs @ 44fps on teensy 3.2

Isn't there a way to set these Ethernet settings without modifying the files?

natcl, 2024-02-14:
Note that these tips are for the w5500 module, not the one that is on the teensy 4

joex92, 2024-02-14:
What's the module on Teensy 4.1?

natcl, 2024-02-14:
It’s built into the processor directly, there’s an alternative library that can be used called QNEthernet, there’s info here:PJRC Storepjrc.com--lecaude.com studioimaginaire.com

joex92, 2024-02-14:
where?

joex92, 2024-02-14:
but how can I make QNEthernet work with artnet?

joex92, 2024-02-14:
and does it work with the Teensy4.1 Ethernet kit?