natcl/Artnet #45: Doesn't work Broadcast

JLightMedia, 2019-04-28:
Hello! I tried to use this library. With unicast mode it works perfect, but when my app share broadcast ArtNet, my Arduino Mega doesn't receive anything. My app provides broadcast ArtNet for x.255.255.255 address. And also, I figured out that other apps that can share full broadcast ArtNet (255.255.255.255) work good with this library. I captured UDP packets from different app and figured out that only Destination IP Address parameters are different. Do you have any idea?

natcl, 2019-04-28:
This is usually related to your network settings in your sketch, have a look at the subnet mask:
https://www.arduino.cc/en/Reference/EthernetSetSubnetMask

JLightMedia, 2019-04-28:
artnet.begin functions has only two variables IP and MAC. And 2.255.255.255 broadcast address enough open to cover even subnet  255.255.255.0. I do not think that the reason in subnet mask.

natcl, 2019-04-28:
You can call the regular Ethernet methods before artnet.begin()
The artnet library is only a parser, the network code is dependant on the Arduino Ethernet library.

JLightMedia, 2019-04-28:
But if call Ethernet.begin method before Artnet method will the same Ethernet method in Artnet library not redefine settings for Ethernet?

JLightMedia, 2019-04-28:
I tried your advice unfortunately it doesn’t work.

JLightMedia, 2019-04-28:
Thank you for your help, I think something bed with my Ethernet library.

natcl, 2019-04-28:
Artnet.begin can be called both without arguments and with Mac and ip as arguments.
If you do not call it with arguments you need to initialize the ethernet library before.
You could test an example ethernet UDP sketch in Arduino's example folder to test out if you can receive broadcast messages.

JLightMedia, 2019-04-28:
Thank you so much! I’ll try it.

MathieuMH, 2019-05-15:
Any luck? I'm struggling with the same problem. I want to drive some WS2811 leds.
Used the OctoWS2811 artnet example. The initTest() function works (I see red, green, blue leds).
I added some serial debug info and the onDmxFrame routine is never called.

my setup:
MagicQ chamsys software on my laptop as ArtNet transmitter
Teensy 3.2 with Wiz850io (tested with the webserver example which works flawless).
DLINK DIR-505 (Laptop on Wifi and Teensy on cable)
Laptop 2.9.200.2
Teensy 2.0.200.15
smartphone with artnet viewer (and it shows there is artnet data on the network).

Tried the above suggestion use the Ethernet library to set the ip and mac. But did not change anything. Either way the Ethernet.linkStatus() == LinkOFF return true. Which means that it appears that there is no cable connected. While there is (green led full on and the orange flashes periodically).

natcl, 2019-05-15:
It doesn't work when broadcasting or it doesn't work at all ?

MathieuMH, 2019-05-16:
Hi, in my situation is does not work at all. By default MagicQ chamsys is set to broadcast.

natcl, 2019-05-16:
Are you able to test with unicast from another source? What ethernet shield are you using?
Also what network settings did you use for your study. It's usually the subnet mask that is no set properly.

MathieuMH, 2019-05-18:
Wel in your example the subnetmask is not defined. In my DIR-505 form DLINK which I regularly use together with my Chamsys lighting desk for live shows, the subnet mask is set to 255.0.0.0 (as in the Art-Net specification). So this means that for artnet the board cast address is 2.255.255.255 right? This is what I had setup in the code. Is the subnetmask of the artnet code set? I don't see a function to change the subnetmask.

I just gave unicast a try with the same software (MagicQ Chamsys). No difference, stil not receiving anything the routine is again not called either. I don't have experience with other artnet transmitters. Can you recommend one for MacOS?

I'm not using one of the default carrier boards. But I made my own based on the OctoWS2811 and Wiz820io adaptor. The design you can find this repository https://github.com/MathieuMH/ArtNetLedPixels
I have to add that it is not quite up to date since, I made two manual changes. For some reason I missed in pin numbering and the reset and chip select were not on the correct pin.
As state before, when I load the webserver example onto my board I can surf to the board en see the analog reeds of the ports as it should. So I'm quite certain that the hardware is functioning.

natcl, 2019-05-18:
You can play with with more network settings by using the standard ethernet library calls, specifically:
https://www.arduino.cc/en/Reference/EthernetBegin

Code would then look like this:

Ethernet.begin(mac, ip, dns, gateway, subnet); 
Artnet.begin()

Notice we call Artnet.begin() without arguments because it is now initialized before.

This is most probably an issue with your network config.
--
lecaude.com 
studioimaginaire.com

MathieuMH, 2019-05-19:
Spend some more time on the unicast matter. After some reading I found where to set my MagicQ software to unicast instead of broadcast. Unicast seems to work, at least the onDmxFrame() is called frequently. I don't have LED output anymore, not even the initTest(). So now there is probably something wrong with the electronics.
Never the less, broadcast isn't working. Don't get why.
The router/AP's DCHP is set to 2.9.200.1 with subnet 255.0.0.0 (DCHP range 2.9.200.10 - 2.9.200.254)
If I'm not mistaken the broadcast address in this situation should be 2.255.255.254 (which is the default artnet broadcast address). When all this is set, the on onDmxFrame routine is not called. When I switch to unicast, it is called. Either the broadcast address is wrong, or there is something wrong with the code.

natcl, 2019-05-19:
Did you set the subnet mask in your code to match the one on your router?

MathieuMH, 2019-05-19:
Yes, using the suggested:

Ethernet.begin(mac, ip, dns, gateway, subnet);
Artnet.begin()

natcl, 2019-05-19:
A subnet mask of 255.0.0.0 seems a bit unusual, did you try something more standard like 255.255.255.0 ?

MathieuMH, 2019-05-19:
it is for regular networking stuff. But for artnet you 'should' use 255.0.0.0 otherwise you limited to 255 devices.
Therefor my gear is default set with this subnet. And I never had troubles with it, until now. :)

natcl, 2019-05-19:
I'm not near a computer right now but a test you could do is to load one of the UDP receive example of the ethernet library and see if broadcast works, this would isolate the issue more.
If broadcast doesn't work with the standard ethernet library then it's probably a setting there, else it might be something in the artnet library although I doubt it since it only does parsing.

MathieuMH, 2020-02-13:
Meanwhile I figured it out.
Regardless off the fact that I had interrupt issue I discovered something about Chamsys (or Art-Net 4 controller in general).
Regardless of the fact that the node is set to broadcast or unicast. MagicQ chamsys only transmits the universe if it listened too.
So if the attached node is not sending the ArtPollReply indicating to what universes is listens. It stops transmitting. I discovered this using wireshark. Universe 1-4 stayed online. But data for universe 5 and 6 stopped after a while. That was the reason why.
I adapted the library to support Art-Net v4. This allows a ArtPollReply for every universe listened. This is easier to implement.

natcl, 2020-02-13:
Indeed Artnet < 4 only allowed 4 universes per node. Feel free to do a PR ! I assume it doesn't break artnet 3 implementations ?

LittleBigGorilla, 2020-03-15:
I'm having the same problem. Unicast works perfectly fine, broadcast doesn't.

MathieuMH, 2020-04-10:
@LittleBigGorilla please read the above. 

@natcl well comp ability should be checked I guess. But the chance that you have a art-net controller with Art-Net versions 3 or lower is quite small. According to the Art-Net docs, version 4 is backwards compatible. But with the different ArtPollReply i'm not so sure.

DrBit, 2022-02-09:
Hi All

An update for those who have the same problem

I had this problem years ago but I ended up just configuring the Unicast setting and forgot about this. Now I'm trying to program a led node that should definetly be listening to brodcast information from my desk (grandMA2).
I was still observing the same behaviour, broadcast messages from the desk using IP 2.255.255.255 awhere ignored by the teensy/arduino code. Using wireshark I could see the packages being send on address 2.255.255.255 but could not receive any of them on the teensy. Otherwise when using the unicast IP 2.0.0.161 (my teensy), packets where received correctly.

As stated avobe by @[MathieuMH](https://github.com/MathieuMH) the solution is to correctly set the subnetMask to {255, 0, 0, 0};

I haven't modified the library to support artnet4, but instead just changed subnetmask. What indeed I've seen is grandma2 when set to broadcastAUTO is detecting my node and setting the unicastIP automatically only when artnetpackets ArtPollReply are present.

I've also tryed chamsys magicQ and it also works correctly.