natcl/Artnet #47: Artnet neopixel sd

Lotsanoiz, 2019-05-05:
How and where can i enter the start record, stop record and play value's  in the neopixel sd sketch ?

MathieuMH, 2020-01-14:
@Lotsanoiz SD card support is not covered in this repository.
You can always make your own implementation, and add it to the examples once you got it working.

natcl, 2020-01-14:
Sd recording is triggered via artnet by setting certain values, is that what you mean ?

Lotsanoiz, 2020-01-16:
Yes, i thought maybe it was possible to enter these value whitin serial monitor to start en stop recording.
I use this sketch now : https://github.com/tangophi/Artnet_DMX_SD_Card
but sometimes it fails for no reason.

MathieuMH, 2020-02-13:
@Lotsanoiz I guess you better open an issue at that repository. Here only the Art-Net lib is supported.