natcl/Artnet #54: on esp8266 mac address is not included in ArtPollReply

mortenthorsen, 2019-12-14:
The missing mac address in the POLL reply poses a problem with some software.

natcl, 2019-12-15:
Would you be able to provide a pull request ? Working in a foreign country and don’t have access to my gear now.

mortenthorsen, 2019-12-15:
Hey, thank you for the quick response. 

tl;dr: No

No, I'm afraid I can't, the code in the libraries are above my level.   I've been trying to to hardcode a mac into the libary, just as a first step sanity check.. but no luck. I can however say that the mac of the ESP's wifi is easy accessible with "WiFi.macAddress", and [https://art-net.org.uk/structure/discovery-packets/artpollreply/](https://art-net.org.uk/structure/discovery-packets/artpollreply/)  specifies the following:
>Mac 
The Mac field is an array of 6 bytes containing the MAC (Media Access Control) address of the device which transmitted this packet. 
The most significant byte is transmitted first.

Furthermore I'm quite new to the whole git experience, so just had to look up pull request ;) 

However I am ready to test out any code you would be able to throw at me... I understand most of what is going on in the code, and I'm pretty sure my failure so far is down to something as simple as placing the data retrieved from "WiFi.macAddress" and placing it in the right format in an usable array.

MathieuMH, 2020-01-14:
@mortenthorsen I've took a quick look at this part of the code. And your observation is correct. The mac address is not set into the ArtPollReply struct.
Out of my head I would suggest to add this to line 34 of artnet.cpp
(assuming you are using Artnet::begin(mac, ip);

`memcpy(ArtPollReply.mac, mac, 6);`

Try is manually by changing this in your Arduino library. I currently have no hardware/setup to verify it. Once you confirmed it I will create the pull request.

MathieuMH, 2020-02-13:
Meanwhile fixed in my fork of the repository. The whole library is updated to meet the Art-Net 4 requirements.
Once Natcl has the time we should merge the two.

LittleBigGorilla, 2020-03-15:
> Meanwhile fixed in my fork of the repository. The whole library is updated to meet the Art-Net 4 requirements.
> Once Natcl has the time we should merge the two.

I tried to use your fork since I like your changes but I'm not able to make it work. When I install it and try the basic receive example my teensy 3.5 just stops working. No serial connection - nothing. I'm new to github and I was'nt able to open a new issue directly in your fork. Is there a way to contact you somehow? @MathieuMH

MathieuMH, 2020-04-10:
@LittleBigGorilla you can send me a private message with what you have ;-) Include a screenshot or some other relevant these that helps me understand what is going on.