natcl/Artnet #61: Has no member named 'setBrodcast'

NorthbySouth, 2022-01-21:
Im trying to use the ArtnetNeoPixle sketch. Once I go to upload the sketch I receive this error message. 
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino Uno"

'using Artnet = class arx::artnet::Manager<EthernetUDP> {aka class arx::artnet::Manager<EthernetUDP>}' has no member named 'setBroadcast'

What can I do to fix this? 
[image]

natcl, 2022-01-21:
Sounds like you might have an older version of the library, can you check in Artnet.h if setBroadcast is defined ?

NorthbySouth, 2022-01-21:
I just downloaded the library today from Readme.md section. Where I can download the most up to date version? I will also check Artnet.h right now.

NorthbySouth, 2022-01-21:
I'm still newer to coding so sorry if this is wrong, but I believe its defined?
[image]

natcl, 2022-01-21:
yes that's strange, do you happen to also have Teensyduino installed ?
There seems to be a conflict of some sort, perhaps 2 versions of the library are installed ?

natcl, 2022-01-21:
I just re-read your message and it seems like you have a different Artnet library installed than mine,
`using Artnet = class arx::artnet::Manager`
That's not part of this library... Did you install a different library at some point ?

NorthbySouth, 2022-01-21:
Yes. I believe there is another Artnet library installed. I will delete it right now and try again!

NorthbySouth, 2022-01-21:
That fixed it sorry for the silly mistake. Appreciate the help!

natcl, 2022-01-21:
No problem, glad to help !

closed 2022-01-21 by natcl